OrcaSlicer/OrcaSlicer #3435: OrcaSlicer crashes (Error code -4020) when trying to print plate with the new Bambu P1S 01.05.00 Firmware

electricar, 2024-01-02:
### OrcaSlicer Version

1.8.1

### OS version

Linux Mint 20.3

### Printer

Bambu Lab P1S

### How to reproduce

1. Update printer to latest FW 01.05.00
2. Slice a model in Orcaslicer
3. Hit "Print plate" 

### Actual results

Orcaslicer 1.8.1. just crashes when trying to send the send the project over LAN.
Orcaslicer 1.9.0. Beta shows this error: 
[image]

It is possible to send only the G-Code file through Orcaslicer 1.8.1. over LAN.
When I try to send the G-Code file through Orcaslicer 1.9.0. Beta over LAN, the program shows no error, it just crashes.

Note:
I am using the printer in LAN only mode.
I am able to use the Home Assistant Bambu Lab integration to interact with the printer. 
The camera stream in Orcaslicer is also not working. It only shows "loading".

### Expected results

Send the project over LAN and start printing as usual.

### Project file & Debug log uploads

[debug_Tue_Jan_02_14_51_49_14396.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13810917/debug_Tue_Jan_02_14_51_49_14396.log.0.zip)
[Screw+Gift+Box_Jayo_100.zip](https://github.com/SoftFever/OrcaSlicer/files/13810919/Screw%2BGift%2BBox_Jayo_100.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

kiwiholmberg, 2024-01-02:
I installed the latest firmware (01.05.00.00) on my P1S. I use "LAN only mode". Orca on Windows 10. 
Here's what i found: 
**Orca Slicer 1.8.0**: Works as well as before. Camera view works, uploading and printing works. SD card viewer doesn't work (which is expected since it's only available in later Orca versions). 
I did have to re-add the printer since it changed access code after the firmware upgrade. Not sure if it's because i went out of and back into LAN mode to do the upgrade, or if it always happens on firmware upgrades.

**Orca Slicer 1.9.0 beta**: 
Always crashes when connecting to the printer in the Devices tab. Doesn't matter if i wait a while (like stated in the release notes). 
Crashes when trying to print or upload gcode.

This is not a bug report, just another data point. And maybe some hope to those who come to the bug tracker before installing the firmware update :)

electricar, 2024-01-08:
Fixed for me in 1.9.0.
Thank you very much!

closed 2024-01-08 by electricar

eolder, 2024-03-22:
I've run into this same issue occurs with firmware version `01.05.02.00` for the two releases I've tested, `v1.8.0` and `v2.0.0-beta`. Also occurs with Bambu Studio.