OrcaSlicer/OrcaSlicer #3436: Crashing on Generating Infill Toolpath

JohnGames, 2024-01-02:
### OrcaSlicer Version

1.9.0-beta

### OS version

Windows 11

### Additional system information

Processor	AMD Ryzen 9 5900HS with Radeon Graphics           3.30 GHz
Installed RAM	16.0 GB (15.4 GB usable)

### Printer

Bambu Lab P1S

### How to reproduce

1. Open project file
2. Wait 30-60 seconds (due to known bug with connecting to Bambu Labs printers)
3. Slice file
4. Wait for it to get to generating infill toolpath

### Actual results

The program gets to 35% generating infill toolpath then closes the program.

### Expected results

Completed slicing of the file.

### Project file & Debug log uploads

[crash_Tue_Jan_02_10_43_10_0.log](https://github.com/SoftFever/OrcaSlicer/files/13811600/crash_Tue_Jan_02_10_43_10_0.log)
[crash_Tue_Jan_02_10_44_21_0.log](https://github.com/SoftFever/OrcaSlicer/files/13811601/crash_Tue_Jan_02_10_44_21_0.log)
[crash_Tue_Jan_02_10_46_04_0.log](https://github.com/SoftFever/OrcaSlicer/files/13811602/crash_Tue_Jan_02_10_46_04_0.log)
[crash_Tue_Jan_02_10_41_40_0.log](https://github.com/SoftFever/OrcaSlicer/files/13811603/crash_Tue_Jan_02_10_41_40_0.log)
[Launch - Aft.zip](https://github.com/SoftFever/OrcaSlicer/files/13811609/Launch.-.Aft.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnGames, 2024-01-02:
I have reduced the number of parts to try to narrow it down.

If you remove Small Craft - Aft - Engine Mount.stl it slices fine.

If you leave just the Small Craft - Aft -Engine Mount you get a different error before it crashes.
[Infill Toolpath Crash.zip](https://github.com/SoftFever/OrcaSlicer/files/13812547/Infill.Toolpath.Crash.zip)
[crash_Tue_Jan_02_12_29_12_0.log](https://github.com/SoftFever/OrcaSlicer/files/13812549/crash_Tue_Jan_02_12_29_12_0.log)

igiannakas, 2024-01-03:
hm....

[image]

[image]

It doesnt appear as a core slicing issue as I can slice your project on a Mac.

Maybe try uninstalling the programme and reinstalling?

JohnGames, 2024-01-04:
I can slice this on 1.8.1. Followed advice and did a full uninstall and re-install with 1.9 Beta, still crashes.

Possibly platform specific considering the crash occurs due to a DLL?

JohnGames, 2024-01-04:
Resolved by https://github.com/SoftFever/OrcaSlicer/commit/77b6225060fd1b6cbc32865b6397363befd2af1d

Marking as closed.

closed 2024-01-04 by JohnGames

jarodreyes, 2025-05-18:
This is not resolved for me. Crashing at 35% slicing, on Apple M3, 16GB. On BambuStudio 2.0.3.54.

igiannakas, 2025-05-18:
Can you post your saved project file?