OrcaSlicer/OrcaSlicer #3455: Add non-planar slicing through port of abandoned non-planar slicing code from zip-o-mat libslic3r branch

ShaneDelmore, 2024-01-04:
**Is your feature request related to a problem? Please describe.**
I would like to be able to print improved mild top surface curves and slants without the stair stepping produced by current approaches. I believe while there is a good chance there will be models this feature will fail to slice correctly, where it works I think it would deliver results that cannot be achieved by any other method we have available today and far above what can currently be achieved with any existing slicers.

**Which printers will be beneficial to this feature**
Any printer with even a little z-height clearance around the nozzle could benefit from this feature on shallow top surface curves.  Those with more clearance would benefit more greatly on top surface curves and may benefit on overhangs as well.

**Describe the solution you'd like**
A port of the abandoned zip-o-mat non-planar libslic3r work.  A little under 2000 lines changed in the branch.

**Describe alternatives you've considered**
The only alternative I have considered is a post-processing step to add curvislicer.

**Additional context**
The git patch for the branch can be seen at https://github.com/slic3r/Slic3r/compare/master...Zip-o-mat:Slic3r:nonplanar?w=1 with the majority of the functional changes being in NonPlanarFacet and NonPlanarSurface.

ShaneDelmore, 2024-01-05:
Almost forgot the most important item to list under who would benefit, benchies!  Think of all of the little benchies, now with less stepping on the rails and roofs.

davids91, 2024-04-05:
Everybody would benefit from non-planar slicing as curved top surfaces would look infinitely better! 

But I am not sure if the linked branch would be the correct way to go about this. Maybe it would be better to make the issue more generic and target the need for non-planar slicing, instead of suggesting one specific implementation. 

https://www.youtube.com/watch?v=gmePlcU0TRw

davids91, 2024-04-17:
I believe after some digging on the internet that there is currently a copyright/patent issue with non-planar slicing whihc discourages most slicers to actually implement it, this might be a big contributor to this issue: https://www.fabbaloo.com/2019/09/possible-patent-infringement-by-non-planar-slic3r-development

davids91, 2024-04-23:
There exists a Prusa slicer for with non-planar implementation: 
https://github.com/EiNSTeiN-/PrusaSlicer

Summarycan be found here:
https://www.youtube.com/watch?v=cbhWni9f980

davids91, 2024-06-15:
I believe this is a duplicate of #5053

closed 2024-09-21 by github-actions[bot]