OrcaSlicer/OrcaSlicer #3464: Bridges turn into overhang walls with too many walls

zviratko, 2024-01-04:
### OrcaSlicer Version

all?

### OS version

all?

### Printer

X1C

### How to reproduce

1. Download this file https://files.printables.com/media/prints/450075/stls/4243049_50802b66-77ed-4b0f-8105-c8bbaab79e83/neck.stl (or use the attached 3mf file)
2. Slice it with 7 or more walls (0.4mm nozzle)
3. Observe the layer @10mm where slots should be bridged over
4. There are overhang walls instead of bridges

### Actual results

There are overhang walls instead of bridges. Those run in the usual perimeter direction making it almost impossible to print.
I printed this part in the past at least 20 times and only noticed it with PETG-CF now because it got majorly messed up as those overhangs failed horribly with it. I thought it was a bug in newer versions of Orca, but I was able to reproduce it with both Orca and Bambu Studio and PrusaSlicer, including versions I previously sliced and printed with (I went down to 1.5.x verions of Bambu Studio thinking this bug wasn't there), so I probably just never noticed as it still printed fine with PLA (lol).

I believe this is related https://forum.prusa3d.com/forum/prusaslicer/problem-bridge-infill-in-mid-air-when-increasing-number-of-perimeters/, so I think this bug has been there forever(?), but it doesn't manifest with all models and most people don't use that many walls, but with this particular part, that many walls are almost mandatory for a good result.

Let me know if I should file this bug (also) elsewhere (prusaslicer, bambustudio...)

### Expected results

At least the outermost wall should be skipped where needed and bridges should be anchored there, just like with <7 walls.

### Project file & Debug log uploads

[bridgebug.zip](https://github.com/SoftFever/OrcaSlicer/files/13836135/bridgebug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zviratko, 2024-01-04:
I forgot I made some screenshots, but it's probably better to just reproduce it...

[image]
[image]
[image]

zviratko, 2024-04-04:
Still an issue.

KennyVB, 2024-04-14:
still a issue.

closed 2024-07-22 by github-actions[bot]