OrcaSlicer/OrcaSlicer #3468: 1.9.0 Beta:: High CPU Load

CCS86, 2024-01-05:
### OrcaSlicer Version

1.9.0 beta

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

After selecting "print plate", Orca was putting a big load on my AMD 5900X (mostly one core). The print dialog was very unresponsive and laggy. I just needed to rename the print and change the filament slot. The second I shut down Orca, all my system fans wound down.

### Actual results

...

### Expected results

...

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

megavenik, 2024-01-11:
double that, it uses 10% of my 12-core 3900X, even staying still in the background. It also makes the whole windows interface very laggy
[image]

CCS86, 2024-01-11:
Still there in 1.9.0 final

b1u3git, 2024-01-14:
(1.9.0-beta) Similar issue, orca run away process...if that's the correct term.  Even after finishing slice, cpu utilization remains between 5-25%, until program close (term process).  Doesn't occur every slice, has occurred multiple times.   Trying 1.9.0 final now.

yunginnanet, 2024-01-18:
on linux same thing happening (1.9.0 appimage) on big prints, to the point where i can never actually send the print job

CCS86, 2024-03-10:
@SoftFever 

Can you reproduce?

ClubspeedChris, 2024-03-22:
I'm having the same issue, uses 10% of my R9 3900X just sitting in the background with no STLs even loaded.

EU4AX, 2024-07-12:
Ver. 2.1.1 same problems. CPU usage returns to normal if you switch and stay on the device or prepare tab. 
No one seems to have any intention of doing that.. :(

JasonHorkles, 2024-07-19:
It looks like it's trying to refresh the print preview every second or so since you can see the image flicker at times, which makes no sense given that the preview shouldn't change while in that menu

intplus10, 2024-12-11:
Ver. 2.1.1. Same issue, 30-50% load remaining on CPU after slicing. Changing to "Device" tab normalize it to the 0-2% level. Happens with every slice for me.

CCS86, 2024-12-11:
@SoftFever I know you added this to the 2.1 milestones. but we are on 2.2 now and it is still an issue. Any updates?

CCS86, 2025-01-10:
I just checked and it is much better.  OS still holds a pretty steady 5 - 6% CPU on my 5900X rig, with high load on a couple cores, when the print dialog is up. This drops to 0.1% when you close that window.

Felix14-v2, 2026-10-06:
Could you test the latest 2.5.0-nightly? There were many performance improvements implemented recently, and the slicer itself should perform much better now. The only big problem now is the Bambu network plugin, but you can probably avoid its drawbacks by using stealth mode.