OrcaSlicer/OrcaSlicer #3472: Crashing when attemptinmg to save new process

krankydonkey, 2024-01-05:
### OrcaSlicer Version

1.9.0 beta

### OS version

Windows 11

### Printer

Voron Trident

### How to reproduce

Click save on an existing user process, and attempt to rename it to a new preset and save.

### Actual results

The save window will close, but the application will hang indefinitely.

### Expected results

A new user process should be added with the given title.

### Project file & Debug log uploads

[SpeedTestStructure.zip](https://github.com/SoftFever/OrcaSlicer/files/13842642/SpeedTestStructure.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13842651/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jimmyapage, 2024-01-11:
I Have the same issue

flyfr8rs, 2024-02-19:
I also have the same issue.

closed 2024-05-27 by github-actions[bot]