OrcaSlicer/OrcaSlicer #3473: Proposals to improve interaction with the prepare and preview windows

liftbag, 2024-01-05:
**I thought I'd bring together in a single post three modification proposals that would improve, in my opinion, the positioning of the objects on the bed and the slice visualization**

*First proposal: block hidden objects*
 I think it makes sense to block unwanted movements of hidden objects on the bed. I often work with crowded beds with all parts of a project, so that I don't have to save too many 3mf. If the number of objects is large, it is natural that they collide or are interpenetrated. Since I almost always print one at a time, I turn off all the others and just focus on the current object. Unfortunately, in this situation, I have to be very careful not to capture a hidden object with the mouse, so as not to inadvertently change its position.

[image]

*Second proposal: enlarge the axis manipulators*
It may be entirely subjective, but I find the Orca Slicer's tiny axis manipulators rather uncomfortable. I often have to repeat the unsuccessful operation due to not being very precise with the mouse, and in cases of crowded beds, I select (again) a hidden object instead of moving or rotating the object that interests me .

Orca's manipulators
[image]
[image]
[image]

PrusaSlicer manipulators
[image]
[image]
[image]

*Third proposal: lock the Z and single layer slider*
Once the slice has been generated, it is possible to view it layer by layer using the appropriate z-slider. If during the movement the mouse cursor leaves the program window, the slider is released and priority passes to the rotation of the bed. In PrusaSlicer and SuperSlicer, once the z slider is engaged, it is no longer released until the mouse button is released, even if the cursor completely leaves the program window. Again, it could be a subjective preference or due to my lack of precision with the mouse, but when I view the layers of a slice, it is almost certain that the bed will start to do somersaults.

netweaver1970, 2024-01-05:
nice suggestions, I for sure fully support your 2nd and 3rd suggestion, with the 3rd one being my biggest UX gripe indeed.
For the 1st one, couldn't you just use multiple plates in one 3mf project? That way things are more visually and functionally spread out, so easier to select/manipulate/print by themselves? Just an idea...

liftbag, 2024-01-05:
> nice suggestions, I for sure fully support your 2nd and 3rd suggestion, with the 3rd one being my biggest UX gripe indeed. For the 1st one, couldn't you just use multiple plates in one 3mf project? [...]

I agree with you on the priorities. And in fact, I wanted to write that my current workaround to the first proposal is precisely to spread the objects on multiple beds. But for a sort of principled logic, it seems correct to me to also deactivate the possibility of movement if I decide to exclude an object from the slice.

liftbag, 2024-04-07:
The third proposal is still very important in my opinion.

closed 2024-07-15 by github-actions[bot]