OrcaSlicer/OrcaSlicer #3474: Collision Warning Ovals are insufficient for Bambu X1C

DesmoMAS, 2024-01-05:
### OrcaSlicer Version

1.9.0-beta

### OS version

macOS 14.1.2

### Additional system information

CPU: M1 Ultra, 20 Cores
RAM: 64GB
Display Card: Apple M1 Ultra built-in

### Printer

Bambu X1 Carbon

### How to reproduce

Add objects to the print bed.  Set the plate to print-by-object.  Place the objects with sufficient clearance for the anti-collision ovals.  
[image]

### Actual results

When objects are printed, the print head collides with neighboring objects (on the right hand side where the micro LIDAR hangs off) due to insufficient clearance.  Thankfully in this case I was printing thin TPU wipers so the previously printed part simply deflects without disruption or damage.
[image]

### Expected results

Print should complete without collision between print head assembly and previously printed parts.  

### Project file & Debug log uploads

[OrcaCollisionIssue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13843158/OrcaCollisionIssue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DesmoMAS, 2024-01-05:
Bambu Studio collision ovals appear to be wider, in comparison.  I have not tried to replicate with Bambu Studio boundaries but when checking the build plate markings, the Bambu spacing is clearly wider.  Issue is specific to the micro-lidar unit hanging off the side of the print head, as the head would have easily cleared otherwise.  Here's what the same parts in Bambu Studio look like with their anti-collision rings:
[image]

dstulken, 2024-02-26:
If the toolhead is that asymmetrical, I would say that even the Bambu behavior is a bug... or at least, an opportunity for improvement.  You would want the clearance boundary to also be asymmetrical to match the actual physical constraints.

closed 2024-06-03 by github-actions[bot]

HunterzNL, 2024-06-29:
I have had a collision with my print because of this :( . I hope it doesn't effect my printer. 
Orca, please fix this. 

My situation(more info): https://www.reddit.com/r/BambuLab/comments/1dhfmge/print_by_object_collision_orcaslicer/
[image]

[image]
[image]
[image]