OrcaSlicer/OrcaSlicer #3477: Printbed display incorrect

bbinnard, 2024-01-05:
### OrcaSlicer Version

1.9.0

### OS version

WIndows 11

### Additional system information

Display: LG 4K 
Video Driver: AMD Radeon RX550

### Printer

FLSun SuperRacer running Klipper

### How to reproduce

Open OrcaSLicer

### Actual results

[image]

Printbed diagram is offset from printbed background. Previous Beta version did not do this.

### Expected results

Printbed diagram should coincide with printbed background

### Project file & Debug log uploads

This happens with no STL file loaded and does not change after one is loaded and sliced. I didn't think a log file etc. was necessary.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NickSchaf, 2024-01-06:
Seeing the same issue upon opening OrcaSlicer 1.9.0.  Additional information, in case it's of use: Adding a second plate results in the folowing:
[image]

bbinnard, 2024-01-06:
Very interesting. Clearly there is something whacked someplace.

bbinnard, 2024-03-07:
This problem still exists in the nightly build 08 Mar 2024. Is it always going to be like this? Does it happen only for delta printers?

bbinnard, 2024-06-06:
I solved this by making an STL file in the shape of my print bed and using that in the printer definition.

closed 2024-06-06 by bbinnard