OrcaSlicer/OrcaSlicer #3479: OrcaSlicer loads to a blank gray screen, displays Setup Wizard (also a blank gray screen) and then stops

hmhall85, 2024-01-05:
### OrcaSlicer Version

1.9.0 also in 1.8.0 and 1.8.1

### OS version

OS: Ubuntu 22.04.3 LTS

### Additional system information

CPU Intel Core i5@3.5GHZ, 24G Ram, NVIDIA GeForce GT 730/PCIe/SSE2

### Printer

None.  Never given the option to select a printer

### How to reproduce

Download the AppImage 
Set AppImage to execute
Double-click on the AppImage

Also tried building locally  from source with version 1.8.1 and experienced the ssme problem

### Actual results

[image]

### Expected results

The application should start and I should see whatever is to be seen.

### Project file & Debug log uploads

/home/mark/.config/OrcaSlicer/log/debug_Fri_Jan_05_13_02_49_2219389.log.0/home/mark/Downloads/OrcaSlicer-1.9.0/error.3mf

### Checklist of files to include

- [X] Log file
- [X] Project file

hmhall85, 2024-01-06:
I note in the log file that the bambu network plugin is failing to load, and that a required library object is not found.  Further research indicates this could be the cause of my error.  How do I get the bambu plugin to load?  Where do I find the alleged missing library and how do I load the library into the app image, please?
Any chance of me getting an old-fashioned tarball where I can compile my own software outside of the app image infrastructure?

jbroadus, 2024-01-17:
I see this on one of my machines (Fedora 39). In my case, it's related to egl+nvidia. If I remove 
`/usr/share/glvnd/egl_vendor.d/10_nvidia.json`, it works.

jbroadus, 2024-01-17:
You can force it to use mesa: 
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_V1.9.0.AppImage`

closed 2024-04-25 by github-actions[bot]

mehrain, 2024-10-23:
> You can force it to use mesa: `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_V1.9.0.AppImage`

This fixed it for me, im on i3/x11 so i used kmenuedit to set the environment variable after adding "exec_always XDG_MENU_PREFIX=arch- kbuildsycoca6" to my i3 config and reloading i3. 
[image]

LetsTryLibertyAgain, 2025-04-24:
The following worked for me (Ubuntu 22.4, Nvidia GeForce RTX 3050) but it still generated a ton of critical errors when run from the command line:
 
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_AppImage_V2.3.0.AppImage

Thanks!  Now I can import a new printer config file and filament profiles.  The preview rendering is so slow it's unusable, so I'll only force OrcaSlicer to use MESA when I need to import a printer config or filament profile.

Hopefully there will be a real fix for this problem but it's apparently persisted for over a year so I'm not too hopeful.

sanchosk, 2025-05-14:
Hmmm, I am on latest Ubuntu (25.04), tested with Orca 2.2.0 and 2.3.0 AppImage versions.
Both start no problem, but fail to show the objects and the print plate - lots of gtk errors on start.
The EGL VENDOR trick does not for me with nVidia - I end up with 
```
libEGL warning: egl: failed to create dri2 screen
Could not create GBM EGL display: EGL_NOT_INITIALIZED. Aborting...
Aborted (core dumped)
```
Any ideas?

LetsTryLibertyAgain, 2025-05-14:
OrcaSlicer 2.3.1-dev fixed my problem with the blank printer selection menu, or maybe it was a recent Nvidia driver update.  The latest OrcaSlicer seems to be working 100% on my Ubuntu 22.04 installation with an Nvidia GeForce RTX3050 graphics card.

Instead of an AppImage, you might try installing a flatpak.  An added bonus is the flatpak can be easily pinned to the panel.  I'd always had to run the OrcaSlicer AppImages by double clicking the filename in the Files directory.