OrcaSlicer/OrcaSlicer #3487: Support painting and seam painting

Rhoenbussard, 2024-01-05:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Printer

Sovol SV01

### How to reproduce

Import file
choose an object
click Support painting or Seam painting

the object disappears

### Actual results

no possibility to manipulate both paintings

### Expected results

see the object

### Project file & Debug log uploads

[clearance-test.zip](https://github.com/SoftFever/OrcaSlicer/files/13845971/clearance-test.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Sopbeen12, 2024-01-06:
Same Issue here.

kencor1, 2024-01-07:
I had the same problem.  Rolled back to the previous version. Works on it.

davidlaym, 2024-01-08:
I was having the same issue, with a notebook that has Nvidia card and AMD integrated graphics. Setting the OrcaSlicer to use the Nvidia card instead of the AMD IG made it work.

On windows 10/11, this is done in the Graphics settings (Settings>Display>Graphic settings link at the end), choosing the orca slicer executable (at Program Files/OrcaSlicer) and changing the options to use High Performance with the Nvidia Graphics.

nva48, 2024-01-12:
Same problem here on my laptop with an AMD Ryzen 5 4500U with Radeon Graphics. Previous Orca version had no issues. Windows 11.

linyongzheng, 2024-01-12:
I have Nvidia dGPU and AMD integrated graphics. Forcing Orca-slicer to use either integrated or dGPU does not fix the issue of the model not appearing when support painting for me.

Noisyfox, 2024-01-12:
> I have Nvidia dGPU and AMD integrated graphics. Forcing Orca-slicer to use either integrated or dGPU does not fix the issue of the model not appearing when support painting for me.

Make sure you force the dGPU from windows' own display settings, not from Nvidia control panel.

Rhoenbussard, 2024-01-12:
I just test ist on my notebook with Nvidia graphics - all is  ok
The desktop computer has AMD Radeon R7 graphics

cherub-i, 2024-01-12:
I am experiencing the same problem. 
Win11, AMD Vega graphics card.

Forcing Orca to use the on-chip graphics card (Intel "something") did not change the behaviour. 
(But maybe changing that setting did not work as expected - I don't know.)

I had no problems with AMD Vega and previous versions.

PR #3663 (2024-01-15): Fix painting Gizmo rendering on AMD Vega GPUs
Fixes #3395
Fixes #3487

Also fixes an issue that seam & support painting gizmo render improperly when wireframe is enabled:
The wireframe is displayed properly in MMU gizmo:
[image]
But looks like this when switching to seam/support gizmo:
[image]
closed 2024-01-15 by SoftFever