OrcaSlicer/OrcaSlicer #3491: "Fix Model" option gone from menu.

weenis, 2024-01-05:
Describe the bug
I use the "Fix Model" option quite frequently. I've been on the standalone version of OrcaSlicer v1.8.1 since it came out, and I used the feature a few days ago. As of today, it mysteriously disappeared.

I've tried restarting, and I've tried deleting my standalone instance and re-exporting from the archive. v1.8.0 contains the option to "Fix Model".

_Originally posted by @weenis in https://github.com/SoftFever/OrcaSlicer/issues/1802#issuecomment-1877859679_

3mf File for This Bug
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

To Reproduce
Right click on an obect with errors, the menu has no "Fix Model" as it normally does

Expected behavior
Menu contains "Fix Model"

Screenshots

[image]

Printer model

Desktop (please complete the following information):

OS: Windows 10 Pro
Version 22H2

giveen, 2024-01-05:
Have you tried updating to 1.9.0? I'm seeing the Fix Model button with no issues.

weenis, 2024-01-05:
> Have you tried updating to 1.9.0? I'm seeing the Fix Model button with no issues.

I could see the "Fix Model" option in v1.8.1 just fine, until it disappeared. In the previous bug report, it happened to that individual as well. Since I'm using OrcaSlicer in a production environment, I can't be beholden to A) the newest and latest copy, and B) Rolling back and crossing my fingers that the option exists in a previous version.

Nevertheless, since I have an active issue with this in the standalone v1.8.1, and it works in the standalone v1.8.0, I have confidence that it may work in v1.9.0, but this doesn't address the root cause.

There's a bug somewhere, and I'd rather it get resolved instead of washed over.

giveen, 2024-01-05:
I'm not washing it over and I'm no way related to the development team for Orcaslicer, just trying to understand and help. 
I am curious if its the model thats the issue and for some reason, as a bug, the slicer is reacting to it in a particular manner.

weenis, 2024-01-05:
> I'm not washing it over and I'm no way related to the development team for Orcaslicer, just trying to understand and help. I am curious if its the model thats the issue and for some reason, as a bug, the slicer is reacting to it in a particular manner.

I understand, and appreciate it.

I opened many other of my previous projects to see if "Fix Model" was on the menu or not, and in the v1.8.1 it wasn't present. When I opened those same models with 1.8.0, the option was available to me.

giveen, 2024-01-05:
are you able to check if 3d builder is still installed on your computer?  I am curious if the option disappeared because its no longer installed.

weenis, 2024-01-06:
> are you able to check if 3d builder is still installed on your computer? I am curious if the option disappeared because its no longer installed.
I am currently able to see the "Fix Model" option with my v1.8.0 standalone. But it isn't available in v1.8.1

I don't know about 3d builder.

giveen, 2024-01-06:
> > are you able to check if 3d builder is still installed on your computer? I am curious if the option disappeared because its no longer installed.
> 
> I am currently able to see the "Fix Model" option with my v1.8.0 standalone. But it isn't available in v1.8.1
> 
> I don't know about 3d builder.

Okay, then that rules out that as an issue, then. If you had clicked on start and typed in "3D Builder", that would have pulled up the 3d model software that Orcaslicer uses to repair the models, but if you see it in one version but not another then it really is an issue with the program.

Just narrowing things down.

closed 2024-04-14 by github-actions[bot]

djfry538, 2024-12-10:
same issue as the first comment, version 2.2.0 
had it then gone? 
also, I have an issue trying to get printer thumbnails to work.

djfry538, 2024-12-10:
fixed Thumbnails.

KubaMiszcz, 2025-02-03:
so do i it is gone, im not sure if it gone after upgrade or just... gone
> same issue as the first comment, version 2.2.0 had it then gone? also, I have an issue trying to get printer thumbnails to work.

KubaMiszcz, 2025-02-19:
in 2.2.0 is gone again

in bambuslicer works ok