OrcaSlicer/OrcaSlicer #3493: Prime Tower  Klipper M600 has very bizarre behavior and cause Move Exceeds maxium extrusion bug

jonatasfreitasv, 2024-01-05:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Printer

Ender 3 V2

### How to reproduce

- Use default Klipper (oficial documentation) M600 change filamento macro
- Set `max_extrude_cross_section: 6` to prevent Max extruder bug
- Active Prime Tower with default configs

### Actual results

- When Prime Tower start, make a full retraction filament and then runs the M600 by moving the nozzle and carrying many strings of filament with it.
- M600 goes there and does a full retraction again, useless, because there is no more filament.
- On return, the prime tower makes an absurd extruder with a lot of material with X-axis stopped at an insane speed, causing a lot of problems and filament blubs.
- After all this madness, the prime tower decides to work, move the x-axis with care and actually do the prime.

The cause is about configs in Printer -> Single extruder multimaterial parametrs, when zero all configs, all works fine.

### Expected results

By default, disable "Printer -> Single extruder multimaterial parametrs" values / feature.

### Project file & Debug log uploads

printer.cfg
https://cdn.discordapp.com/attachments/1138480666657443880/1192732419778940988/printer.cfg?ex=65aa25a0&is=6597b0a0&hm=c3a1e9a528c19b76e23fa6259298676bce4d6b4644cf284ffb2d525be98a377e&

gcode with bug (CP TOOLCHANGE LOAD section
https://cdn.discordapp.com/attachments/1138480666657443880/1192731671242473482/prime_tower_bug.gcode?ex=65aa24ee&is=6597afee&hm=a908a1061b0b6f2129c6c09fec5941b6b17397f386b0349b98bbbf4db3f06922&

### Checklist of files to include

- [X] Log file
- [X] Project file

Pardiman, 2024-01-13:
The same happened to me.

I looked for the "CP TOOLCHANGE LOAD" lines for each filament change in the gcode and removed the following lines:

```
M600
M106 S0
; MANUAL_TOOL_CHANGE T0
; filament start gcode
SET_PRESSURE_ADVANCE ADVANCE=0.03; Override pressure advance value

G1 F12000
G1  X265.001 Y245.887  
G1 E4 F2100     <---------------------- Remove this
G4 S0
; CP TOOLCHANGE LOAD     <------------- Search this
G1 E18.0000 F180     <----------------- Remove this
M73 P3 R20
G1  X206.376  E63.0000 F1563     <----- Remove this
M73 P3 R19
G1  X265.001  E9.0000 F1094     <------ Remove this
G1  Y246.387 
; CP TOOLCHANGE WIPE
G1  X206.001  E3.2299 F891
G1 F12000
G4 S0
G92 E0
; CP TOOLCHANGE END
;------------------
```

The X coordinate varies, but you can be guided by the exaggerated extrusion.

I didn't find any way in OrcaSlicer to disable this option.

Hope this can help you.

Pardiman, 2024-01-13:
Sorry, I didn't see that you were showing the solution to me:

> The cause is about configs in Printer -> Single extruder multimaterial parametrs, when zero all configs, all works fine.

Thank you.

Haimana, 2024-01-15:
Thank you, it works!

closed 2024-04-23 by github-actions[bot]

AidanHost, 2024-04-28:
not stale, definitely still an issue

@jonatasfreitasv which config values did you set to zero?

gabrielwong1991, 2024-07-22:
Hi I am having the same issue. Any idea how to fix that?