OrcaSlicer/OrcaSlicer #3494: Some hacky manual timelapse support

piotrb, 2024-01-05:
**Is your feature request related to a problem? Please describe.**

Basically I'm asking for timelapse support for klipper. "Support" is a big word here .. I basically just want to be able to toggle timelapse support with a checkbox on the printer profile and perhaps in the print profile.

When using this system: https://github.com/mainsail-crew/moonraker-timelapse .. the only slicer setting needed basically is a bit of gcode that runs inside "before layer change" .. so if this support could break out a separate textbox for "timelapse layer snapshot" gcode .. where that could be put into ..  and then the toggle for timelapse would just either include this gcode before layer change, or not .. 

Down the line, you could potentially actually support the apis to fetch the timelapses .. but this is not needed to start since Fluidd does a good enough job here.

The issue isn't so much that there is no way to do this (there is, just add the gcode to the before layer change). The issue is that i'd like to be able to easily toggle if a timelapse should be made for a print.

**Which printers will be beneficial to this feature**
Voron/Klipper based printers

**Describe the solution you'd like**
I kind of sketched that out in the first block .. happy to hash any details out in this ticket

**Describe alternatives you've considered**
the main alternative is just adding this to the before layer change, and removing it as needed .. but that's quite cumbersome

**Additional context**
Add any other context or screenshots about the feature request here.

Ocraftyone, 2024-01-07:
Should already exist. There is a field in machine gcode for time lapse gcode

pmbroth, 2024-01-12:
where do you insert the time-lapse macro call code, as there is only a post processing script area, and nto before each layer script or post layer script area tpo enter copde.  I only see post processing,

galadriann, 2024-01-13:
I was looking for this too .. in the Printer G-code along with the start and end G-code there is section called timelapse G-code..  handy..

piotrb, 2024-01-19:
Cool so the timelapse gcode is great, but it seems like unless you perhaps have a "supported" printer the option to toggle time lapses (Special Mode: Timelapse) .. which I'm assuming is the toggle .. is not visible .. what controls the visibility of that option?

galadriann, 2024-01-19:
I may be mistaken but that option is I only available on the bambulab version.
There is currently no toggle. You need to go in your printer settings and add the gcode there. Also ideally you need to add code to move the print head out of the way.  I have a slinger bed so had to ensure the bed is also always at the same spot.
We would need 3 options then. 
- enable Timelapse. 
- camera side (left or right). You don't want to the head in the way
- slinger bed or not (in which case the position would need to be place in half of the bed size)

The is the code I added my my printer. Camera is on the left side

G1 F9000 X190 Y100 ; position head on the right. Bed half course
M400 ; wait for move to complete
TIMELAPSE_TAKE_FRAME

While we can add the code manually it would I deed be amazing to have this as a feature

piotrb, 2024-02-02:
I've been using https://github.com/mainsail-crew/moonraker-timelapse .. it seems to support telling it where to move the head automatically, so then you just need `TIMELAPSE_TAKE_FRAME` .. the rest you can configure through the UI config that shows up in Fluidd (no idea if mainsail supports configuration of timelapses as well or not)

closed 2024-05-10 by github-actions[bot]

heitorPB, 2024-08-30:
Did you guys figured out how to add the gcode to each layer change? Is it 1 command to add to the gcode file or need to add `TIMELAPSE_TAKE_FRAME` multiple times?