OrcaSlicer/OrcaSlicer #3496: How to hide G-code preview?

kozross, 2024-01-05:
I just updated to 1.9, and I now get a G-code preview in addition to the usual slicer preview:
 
[image]

How do I turn this off?

amdavidson, 2024-01-05:
I have seen this before from time to time in previous versions as well and also do not know how to hide it. 

Am I missing something?

WluhWluh, 2024-01-05:
Press the C key to disable or enable G-code preview.

amdavidson, 2024-01-05:
Ah, should have checked help->keyboard shortcuts hadn't looked there. Thanks @WluhWluh

unrushed-boots, 2024-01-05:
https://github.com/SoftFever/OrcaSlicer/issues/3489 same issue. seems this was turned on by default in 1.9

tlhintoq, 2024-01-06:
Just look in the menus for Pete's sake.

[image]

bistory, 2024-01-06:
> Just look in the menus for Pete's sake. [...]

This is the right answer, @kozross please don't forget to close this issue.

closed 2024-01-06 by kozross