OrcaSlicer/OrcaSlicer #3497: Allow editing compatible printers for filament and process profiles more easily

kozross, 2024-01-05:
**Is your feature request related to a problem? Please describe.**

Currently, the only way to edit what printer(s) a profile is compatible with is to:

1. Export all your profiles (and you can't do just one).
2. Manually edit the `compatible_printers` section of the JSON-formatted file.
3. Re-import the modified profile.

This is particularly salient when you make 'derived' profiles from manufacturer stock ones: you retain all of their 'generic' filament profiles, and anything you make atop of that continues to be associated with not only the stock manufacturer profile, but typically _all_ stock manufacturer profiles. This editing is a bit tedious and involves jumping back and forth between the slicer and a text editor.

**Which printers will be beneficial to this feature**

Any time you need a 'derived' profile and want to be organized.

**Describe the solution you'd like**

`compatible_printers` should be an editable field within the profile itself, while in the slicer.

**Describe alternatives you've considered**

The current workaround of exporting, manually editing a text file, then re-importing works, but it's tedious and time-consuming.

**Additional context**

PrusaSlicer doesn't have this problem: you can edit that field in it just fine.

Zaunei, 2024-04-08:
Editing the files in the config folder also works, but it would be really handy if you could set the compatible printers in the slicer.

kozross, 2024-04-12:
@Zaunei - the weirdest part about this is that PrusaSlicer already has the ability to do this. It probably got removed somewhere along the way (BambuSlicer would be my guess).

Gwoony, 2024-04-19:
I could not change it Manuel. If I do this in the config folder, the setting will be overwritten with the next start of the slicer. How could I do this?

kozross, 2024-04-19:
> I could not change it Manuel. If I do this in the config folder, the setting will be overwritten with the next start of the slicer. How could I do this?

You have to export, modify, then re-import, like I described above. This is tedious, and one reason I opened this issue.

Fragmon, 2024-05-11:
Whats the actual status. My Filamentlist ist anoying because there are 200 filaments for 4 printers saved and i see all on every printer.

kozross, 2024-05-12:
> Whats the actual status. My Filamentlist ist anoying because there are 200 filaments for 4 printers saved and i see all on every printer.

Given that we had to un-stale this issue before any of the maintainers commented? Probably not getting improved any time soon.

kozross, 2024-08-11:
Not fixed yet, so once again un-staling.

john-f-chamberlain, 2024-09-20:
I too would like to be able to be able to modify the "compatible_printers" field from within the slicer.
I've got 3 different printer models and having to set up every filament 3 times is rather annoying.

closed 2025-01-13 by SoftFever

clingmann, 2025-07-10:
Filament management on this level is needed so much, it is so stupid that you can't share a filament profile across printers. Please correct this LONG standing issue with the slicer. I mean even the ability to see the other printer profile and "inherit" from it for a new one would be helpful. Why force everyone into the text editor. Similarly this creates issues moving filament profiles between slicer variants.

PR #6235 (2025-01-13): enabled dependency setting for filament and process
# Description
Enabled dependency setting for filament and process for **Developer mode**. Dependency feature seems to be commented out by BBS, this PR is just to uncommenting dependency tab related code. This fixes #1655  *This is probably not release-ready, but I want to share what I have.

Only changes I added were
from `toggle_options();`
to
`    if (m_active_page && !(m_active_page->title() == "Dependencies"))
        toggle_options();`
This is necessary to prevent app crashing from viewing Dependency tab.

edit: also fixes #3497 

# Scre [... cut]