OrcaSlicer/OrcaSlicer #3498: fan speed up time in printer seeting is not working

jlo1975, 2024-01-05:
### OrcaSlicer Version

1.8.1

### OS version

win10

### Printer

bambu lab P1S

### How to reproduce

1 use the defaut bambu la P1S defaut printer setting, 
2 create a new project and add the autodesk  FDM test (in primitive popup menu)
3 save the gcode generated (we are focusing on the overhang part)

4 modify the printer setting and put 1 second to fan speed up time
5 regenerated the gcode

perform a diff on both gcode and you will notice even if the option is active, no special gcode is generated top handle the fan speed up time (no barrier, no wait G4 code, no slow down of the extruder).....

see screenshot below and generated gcode (you will notice some noise with the M73 not generated at the same line)
You will also notice some M106 fan speed command now missing when setting the fan speed up time is 1 second.

for the long story about this bug with more test see -> https://github.com/SoftFever/OrcaSlicer/issues/3350
where a first i asked to get this feature (I didn't it was implemented but seem now to be bugged)

default setup
[image]

with  1 seconde on fann speed
[image]

kdiff3 result on the 2 generated GCODE
[image]
[image]

### Actual results

the fan speed up time is not handled in the GCODE

### Expected results

see some barrier / wait command or slow extruder speed alt/slow down

### Project file & Debug log uploads

here the gcode for both test 
[gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13847072/gcode.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

maziggy, 2024-01-09:
Same here with the latest version on MacOS.

jlo1975, 2024-02-15:
ok I found out the problem is coming from the check box -> only overhangs...

here the test result:

a test to replicate the problem
[image]

THE BUG !
[image]

AND FINALLY !!!!!
[image]

jlo1975, 2024-02-15:
on the last screen shot you can see that the fan part ramp up way before the critical parts.
this is what we wanted!

the checkbox only overhang is missleading as it implies to limits the ramp up time frame ONLY inside the overhangs time frame.
So you can't have any fan ramp up just before entering the overhang part....

the side effect of untick this check box is the rampup will also applied on part that are not overhangs.
But it is a minor problem vs overhangs to be cooled correctly (Yes I know i'm unclear sorry......)

closed 2024-05-23 by github-actions[bot]