OrcaSlicer/OrcaSlicer #3500: Slicer crashing when switch printer since adding the A1

win10ha, 2024-01-05:
### OrcaSlicer Version

V1.9.0 Official Release

### OS version

Windows 10

### Additional system information

I7 4770K
16 GB DDR 3
GTX 1070 TI

### Printer

Bambu A1/ Ender 3/ CR10's

### How to reproduce

Switch printer profiles from Bambu A1

### Actual results

Hard crash no errors reported, seems to work ok if I remove the Bambu A1 profile (crashes when removing)

### Expected results

The ability to switch between profiles 

### Project file & Debug log uploads

[Orca Crash.zip](https://github.com/SoftFever/OrcaSlicer/files/13847095/Orca.Crash.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

win10ha, 2024-01-05:
Also happens with Bambu Slicer

[image]

[image]

Egosumumbravir, 2024-01-06:
I've been getting crashes just like that switching from Ender3 to Bambu X1C profile on the 1.9beta.

closed 2024-04-14 by github-actions[bot]