OrcaSlicer/OrcaSlicer #3502: Advanced costing for print jobs.

EdwardChamberlain, 2024-01-05:
**Is your feature request related to a problem? Please describe.**
Currently the print cost shown on the preview page is (I believe to be) a simple calculation of filament cost multiplied by the amount of filament used. This provides a great estimate of material cost for the print but does not provide an accurate cost of the print job as it does not include some key costs:

- Energy
- Print time costs
- Printer depreciation
- Load / unload fixed costs

The current cost estimation provides an inaccurate measure of true printing cost.

[image]

**Which printers will be beneficial to this feature**
All prints. 

**Describe the solution you'd like**
Allow users to enter extra metrics about their machine for fixed costs and running time costs. Fixed costs can be used to cover printer loading and unloading / packaging / slicing / setup etc while running time cost can be used to estimate energy use, depreciation, maintenance, etc costs.

Total Estimation screen will then show the extra metrics like so:
Material Cost: 5.75
Running Cost: 17.50
Setup Cost: 3
Total Job Cost: 26.25

**Describe alternatives you've considered**
Currently this calculation is typically done by copying print time and material use figures to an external spreadsheet but this requires manual steps to extract slicer data making it slow and prone to errors. It is also unlikely that average users are completing these extra steps for standard prints leaving them without a realistic understanding of cost. 

**Additional context**
Add any other context or screenshots about the feature request here.

tlhintoq, 2024-01-06:
Does any slicer have this stuff?  I've never seen any slicer try to be your shop management software.  To be fair your "true cost of production" fails to account for a lot of other things: 

- Per square meter of building expense
- Air con to combat the heat of the machines
- Indirect labor costs that are still needed for the shop, like cleaning and maintenance.
- Materials handling and inventory
- blah blah blah

All of that is just not part of a slicer's job - In my personal opinion.

Personal two cents worth based on my little print farm of 14 FDM and 8 resin machines - might not be even close to your farm...

Trying to micro-estimate like that has always been (for me) a case of spending $5 in micro-accounting to track 75 cents of costs.

Generally speaking if you're trying to micro-price like that you're not giving yourself near enough margin.  All it will take is one bump in filament cost, a few failed jobs, whatever, and you're in the red.

Most shops will figure out their operating costs - the cost of just being in business, covering everything from rent to air conditioning, benches and racks, tools and consumable supplies - everything that is overhead - can calculate a base per-hour cost just to break even.  Then double that to account for dead time.  If you're not operating at at-least 50% capacity you have bigger issues than if your slicer is calculating job costs.

Once you know that your billable cost-of-doing-business is $5/hour then you don't need any more from your slicer than what you're already getting: Materials cost and time.

EdwardChamberlain, 2024-01-06:
> Does any slicer have this stuff?

Not that I am aware of. Cura has a similar cost estimation to the slic3r forks. I'm not sure that what others are doing is relevant though. 

>  I've never seen any slicer try to be your shop management software.

I dont believe that this is trying to replace your shop management software. Those running shop management software will obviously benefit from using those tools but many users (I imagine most) are not using these packages and therefore have very limited visibility of the real world costs of using their machine in their home. What you have listed are very clearly business expenses and much less directly related to the printer itself. 

In a similar vain you could take the argument the other way and ask why any slicer asks for filament cost. The slicer should not be managing this data either and should only be providing a mass / volume of material used for the shop management software to link to material cost independently. I think the obvious answer here is that both tools can calculate the same things given the right inputs, it is more about making this data available to a user in a meaningful way / place and catering basic functionality to those that do not have additional software.

What you have said is all great business advice but a large number of users are not businesses. This feature request comes from my experience where I have seen a lot of hobby users come and say "this print costs me $0.15!!!" because that is what their slicer has told them that the print will cost but they have failed to account for depreciation of their machine, energy costs etc, that only add to the cost. Having a box under the printer settings that works exactly the same as the "filament price" box but multiplied by time rather than weight would allow rough approximation of a more realistic cost for those users who are printing as a hobby. Perhaps these people want to offer to print something for someone and want to know roughly how much to ask for without having to open an external spreadsheet, work through the numbers etc - this is a tool for them.

[image]

tlhintoq, 2024-01-06:
> What you have said is all great business advice but a large number of users are not businesses. This feature request comes from my experience where I have seen a lot of hobby users come and say "this print costs me $0.15!!!" because that is what their slicer has told them

Never going to educate the 99% of hobbyists that simply aren't interested in learning or an understanding.  Just spitting in the wind there.  Might as well be trying to educate them on why PLA isn't the material for everything... or try to teach them why they need to *understand* this hobby and not just ask "what settings do I need?"

The user that you could teach things like overhead, cost of operation etc., already have an IQ greater than their shoe size and probably already grasp the basics of real cost and hidden expenses: Most adults should based on having run across it in 100 other aspects of their lives.

The user that you need to teach these things to aren't going to accept it because of a box on screen - aren't going to set up the slicer for 10 more values anyway: They barely want to set it up enough to get a good print - they certainly aren't going to set up values like the cost of electricity: Which will vary depending on night v. day per KW cost anyway in a lot of place... etc.  They just won't.  Half of them are teens running Bamboo for turn-key, no learning printing of flexi lizards.

Basically, those you can educate don't need  you to.  Those that need you to, don't want to hear it anyway.

EdwardChamberlain, 2024-01-06:
This is a very negative and disappointing view to take of the 3D printing community.

This is a feature that would be directly beneficial to me, and many others that I have spoken to about it. Thus a feature request was raised. It is clearly not a feature that is of interest to you but that is ok - you do not need to use it.

cochcoder, 2024-03-12:
@EdwardChamberlain Does my PR #4392 solve your feature request?

EdwardChamberlain, 2024-03-17:
@cochcoder Yes! That looks like exactly what I was looking for.

cochcoder, 2024-03-17:
Great!

cochcoder, 2024-06-17:
Still a work in progress

cochcoder, 2024-09-16:
Need to come back to this at a later date

tlhintoq, 2024-12-16:
Is this ever going to get put in?  Wasn't this code finished months ago?

cochcoder, 2025-01-08:
> Is this ever going to get put in? Wasn't this code finished months ago?

I did finish some rudimentary code months ago, the issue was that I couldn't figure out a way to make it accurate enough that the user could trust it to be correct. 

If anyone has any ideas for how to make it accurate enough and can help confirm its accuracy, I'm all ears!

tlhintoq, 2025-07-10:
Still hoping that some day this gets done

cwfanion, 2025-07-20:
I too would like this feature, as I grew used to it with Cura, but have since switched to Orca. For Cura, I used this GitHub project: https://github.com/YiamiYo/Cura-PowerCostPlugin. Hopefully his release helps answer your pending questions on fixing this.

EdwardChamberlain, 2025-07-20:
I think the most straightforward approach is to break down total cost into three components:
- Material cost – already implemented: filament used × cost per kg
- Fixed cost – a static value per print to account for handling, setup, etc.
- Running cost – calculated as print time × user-defined rate (e.g. to cover machine usage, energy, depreciation)

The total cost would be the sum of these three.

I don’t think it’s necessary to model energy use, depreciation, or other specifics directly. Letting users define fixed, time-based, and material costs provides a simple but flexible system, allowing them to roll in any additional overheads as needed without shifting too much responsability onto the slicer to be a complex costing model.

tlhintoq, 2025-07-22:
Two of those three are already in place.

- Material has a cost per unit.
- Machine has its running cost per hour

Handling cost per batch… Meh… Varies too much depending on who's doing the work.  If its me that's one hourly rate but if its one of the workers/interns etc. that's a different pay rate.  Not to mention at some point the cost of micro-calculation becomes more than the cost of the actual labor.  You see companies do that all the time where the accountant's $50 overshadows the $5 is losses they found.  Or how inventorying 15 sizes of boxes is more costly than the 'waste' of  only keeping 3 but using a big box for small things: You see Amazon using this reality all the time.  

Personally, just one man's thoughts... If your profit margin is so small that you have to track the 3 mins of labor per batch then you're already hosed.  Because it also means you're already in trouble for the 50cents in cleaning alcohol not on this list, or the shop overhead of air con for the 3 hours the job ran, or the 4% loss from print failure rate, or the 3 hours this week the employee was idle because you didn't make enough sales to keep them busy while on the clock.  At some point you just take the small stuff and put it in the same bucket of "overhead cost of doing business" and weight all the costs for the month against all the profits.

tlhintoq, 2025-10-22:
Didn't someone already do all the work on this?  I seem to remember there was a branch for it and everything.
Why is this now `stale` and not integrated to the program?

thelegendtubaguy, 2026-10-28:
With the advent of the plugin system, this is likely better left as a plugin.

EdwardChamberlain, 2026-10-28:
Agree

closed 2026-10-28 by EdwardChamberlain