OrcaSlicer/OrcaSlicer #3507: App shuts down while using the text tool

energizer1389, 2024-01-06:
### OrcaSlicer Version

1.9.0

### OS version

Mac mini M1 - Mac OS Sonoma

### Additional system information

Application shuts down when I try to add or edit using the text tool

### Printer

Bambu P1S

### How to reproduce

1. New project
2. Add model
3. Click on text tool
4. Type tex (Most times crashes here)
5. Move text around (It crashes here at times too)

### Actual results

That application closes

### Expected results

I should be able to continue without application closing automatically

[tutk_log_avapi_2606.txt](https://github.com/SoftFever/OrcaSlicer/files/13848053/tutk_log_avapi_2606.txt)

### Project file & Debug log uploads

[Hand holding heart.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13848038/Hand.holding.heart.3mf.zip)
[tutk_log_avapi_2606.txt](https://github.com/SoftFever/OrcaSlicer/files/13848053/tutk_log_avapi_2606.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-01-06:
Maybe a Mac thing or a font issue?
I can't get it to crash on Windows with a variety of fonts, sizes, rotations or positions.
[image]

energizer1389, 2024-01-06:
It crashes every single time. I don’t even change the default font.

JV

energizer1389, 2024-01-07:
I have tried again today. I think this is related to that Mac Bug fix for text emboss with complex shapes. I think it needs more work. I made a few more tests today on a second M1 Mac I am experiencing the same issue.

GroteGehaktBal, 2024-01-07:
I'm having this same issue on a Macbook Pro M1. Happens every time when using the text tool. I tried using different fonts but still crashes the app.

clemy49, 2024-04-03:
Similiar issues here. The text tool is going crazy. Especially when I try to copy text into the text input box it copies multiple times or it closes the editing mode.

tlhintoq, 2024-04-03:
FYI: I can't get it to misbehave (yet) on version 2.0 stable release.

Lazersfly, 2024-04-03:
I have experienced this crash pretty consistently since the introduction of the emboss tool, even on the latest stable 2.0.0 release. I've tried the standard troubleshooting steps (reinstall, wipe of user profiles, etc) with no success. Have also been able to reproduce the crash on another identical machine with no prior install.

To note: I can't seem to reproduce the crash using some files, while other files crash with any use of the emboss tool. This is just speculation, but given that stls with non-manifold edges seem to always be effected, this might be a bug within the rendering code for the emboss tool. Just a guess though.

My setup and steps to reproduce below -
2023 M2 Max Macbook Pro
MacOS 14.4.1
OrcaSlicer 2.0.0

1. Import affected stl
2. Add text using emboss tool
3. Start clicking up/down arrows for the depth setting, eventually (usually within 5-10 adjustments) the entire program will crash to desktop and throw an "app closed unexpectedly" message. I believe this crash can happen with any setting change within the emboss menu, but I've been using the depth setting for testing.

dcs3473, 2024-04-25:
I also experience crashes every time I use the Text Tool.   The quickest way for me is to choose Advances and try and click on Surface.  It fails every time one Mac Mini and Macbook Pro 14.

closed 2024-08-01 by github-actions[bot]

TammyWalker-glitch, 2024-10-27:
Orca Slicer 2.2.0-rc
Windows 11
The program crashes when I select the emboss tool