OrcaSlicer/OrcaSlicer #3508: Ability to re-organize plate order

tlhintoq, 2024-01-06:
It would be of tremendous help to be able to re-order the plates of a job.

- Sometimes its nice to move similar objects in sequence
- Move completed plates to the end so you know what's next or where you left off printing
- - Or maybe just a clickable check box on the right-side tools to show `[x] Done`.  I feel silly using the `Lock` tool right now for this indicator.
- Re-organize once you slice and know the print time.  For example: Two 4hrs jobs making up one 8hr shift of work, followed by a 12-16hr job to run overnight.  Putting the jobs in workflow order would do a lot for job sequence planning.

[image]

Glenno-H, 2024-01-10:
being able to reorganise plate order would be soooo nice.

tlhintoq, 2024-01-17:
Keep alive - Still an issue in 2.0 developer/nightly

tlhintoq, 2024-02-08:
Hey bot, please don't close me - this is still an issue.

tlhintoq, 2024-03-13:
Problem still exists in current 2.0 nightly as of 13mar2024

tlhintoq, 2024-04-15:
Keep alive.  Problem still exists in current 2.0 nightly as of 15apr2024

Moeser, 2024-05-03:
I'm currently working on a project with over 20 build plates, and being able to re-order the build plates would help me keep the project more organized and print in the correct order.  For doing just-in-time printing tasks for larger projects, this would help make sure the correct parts are printed when needed.

tlhintoq, 2024-05-10:
Bot nudge.

stevenhu70, 2024-05-10:
ohhhh yes please!!!   I have a large project with 10+ plates and it would help sooo much if I could re-order them

Shadowpheonix, 2024-05-19:
This would be a very nice enhancement.

geekypilot, 2024-05-20:
This feature is definitely needed for larger projects.

emeyedeejay, 2024-06-03:
June '24 bot bump

+1 from me 🙏

mgjv, 2024-07-04:
I would also like this. I have several large projects that have become messy over time, and plate reordering would help me work with them a bit more easily.

malteschulz, 2024-07-08:
This would be very helpful for keeping track of a large project

tlhintoq, 2024-07-30:
@Azio-Pantheon 
I very much appreciate the work you did to give us the ability to `move plate to the front`.  While not a total ability to re-org the entire project it should do a lot for workflow tracking and ordering.

- Select - move to front
- Select - move to front
- {repeat}

Eventually you have the order you'd like to do the project in.
Or just select - move to front, tells you the most recently printed part.

Thanks!

tlhintoq, 2024-08-16:
> [Azio-Pantheon](https://github.com/Azio-Pantheon) linked a pull request [3 weeks ago ](https://github.com/SoftFever/OrcaSlicer/issues/3508#ref-pullrequest-2432989746) that will close this issue

3 weeks ago.  Not yet merged.  Just thought I'd throw a nudge in there for this.
cc: @SoftFever

MattiaConfalonieri, 2024-09-04:
+1 for this feature

MKloberg, 2024-09-16:
Me too! This would be awesome.
My use case is 37 different ear ring models that each need a specific color/filament, I need to keep in alphabetical order to stay on top of all the stl's.
Then I would like to insert production plates at the top/start/beginning to sort these into material based plates, to group them by filament/color.
This would be a major enhancement for us commercial printers and super cool, even in a crude way so we can move the plates around (doesn't have to be drag and drop), some kind of index field in the plate properties to rearrange them would be fine.

[image]

MKloberg, 2024-09-16:
I just ran into another thing that makes no sense. There's a hard limit of how many plates can be created at 36 plates.
Can we remove that limitation?
Why not? Is there a reason for that?
Sure, the users machine will blow at some point, but what if there is ample CPU/GPU available to lean that far out?
To organize my jewelry project properly, I will need at least 10 more plates to make it work...
Thank you for looking into this also.

[image]

tlhintoq, 2024-09-16:
> I need to keep in alphabetical order to stay on top of all the stl's.
What a fraking nightmare.  Just a thought-Use the plate names?  Or put the color on the plate?
[image]

MKloberg, 2024-09-17:
I found where the plate limit is defined (hard-coded) if that helps.
Would it be possible to make that a preference item - defaulting to 36, so we can crank it up if needed?

[image]

closed 2024-09-25 by SoftFever

hermitdafunk, 2025-04-30:
+1, please.

dsluo, 2025-05-29:
I don't really feel like #6262 is a complete solution for this issue. Moving to front is nice, I guess, but it still doesn't let you reorder plates arbitrarily without manually shuffling them around.

tlhintoq, 2025-05-30:
@Noisyfox  Should folks like @dsluo and I keep making suggestions here even though this ticket is closed?  Does that get any attention or would a new ticket be better?

I agree that more ability to re-organize would be nice.  I kinda thought the `move to first` was a fast stop-gap measure until a more complete solution could be completed.

Here's a thought that shouldn't take any new UI, no new widget tools or mouse behaviours.  Let us grab the plate itself from the object list and move the entire plate up and down between other plates, exactly the same way we grab an object *on* the plate and move it to change sequence.
(also, a `collapse all` button at the top of the plate list would make this a ton easier)

[image]

scrith, 2025-08-14:
+1 to this feature.

I sometimes create new revisions of models, but need to retain the current revision. It would be really awesome if I could make a new plate with the updated print and then move it to the workflow space it belongs at.

PR #6262 (2024-09-25): Feature/move plate to the front
# Description
Added a button on the plates to move the plate to the front. Gives users the ability to re-order the plate to some extent. It resolves #3508 

# Screenshots/Recordings/Graphs
[image]

## Tests
Tested locally on Windows 11 system.