OrcaSlicer/OrcaSlicer #3509: Broken UI for config export

kozross, 2024-01-06:
### OrcaSlicer Version

1.9.0

### OS version

Arch Linux

### Additional system information

CPU info according to `cat /proc/cpuinfo`:

```
processor	: 23
vendor_id	: AuthenticAMD
cpu family	: 25
model		: 33
model name	: AMD Ryzen 9 5900X 12-Core Processor
stepping	: 0
microcode	: 0xa201016
cpu MHz		: 2200.000
cache size	: 512 KB
physical id	: 0
siblings	: 24
core id		: 13
cpu cores	: 12
apicid		: 27
initial apicid	: 27
fpu		: yes
fpu_exception	: yes
cpuid level	: 16
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba ibrs ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 erms invpcid cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local user_shstk clzero irperf xsaveerptr rdpru wbnoinvd arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif v_spec_ctrl umip pku ospke vaes vpclmulqdq rdpid overflow_recov succor smca fsrm debug_swap
bugs		: sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass srso
bogomips	: 7402.89
TLB size	: 2560 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]
```

GPU info according to `lspci`:

```
0a:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] (rev c1)
```

### Printer

Qidi XPlus-3 (custom)

### How to reproduce

1. Go to File-> Export -> Export Configs.
2. Observe the window and notice it's unreadable.

### Actual results

Screenshot pretty much speaks for itself:

[image]

### Expected results

A usable window, with all text rendered.

### Project file & Debug log uploads

[export-configs.zip](https://github.com/SoftFever/OrcaSlicer/files/13848427/export-configs.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

el-quique, 2024-01-06:
The same in linux mint debian edition

jjramsey, 2024-01-07:
I have a probably related problem. "Export configs ..." causes OrcaSlicer to segfault. In my case, I have an NVIDIA card: `NVIDIA Corporation GA106 [GeForce RTX 3060 Lite Hash Rate] (rev a1)`. Also running on Arch Linux, with OrcaSlicer version 1.9.0.

closed 2024-04-14 by github-actions[bot]