OrcaSlicer/OrcaSlicer #3514: Flickering window when sending a job to printer

NordicRavn, 2024-01-06:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10 Pro

### Additional system information

CPU: Inter i5-8400, RAM: 16GB DDR4, GPU: Nvidia RTX 2060 

### Printer

Bambu Labs X1C

### How to reproduce

1. After slice go to Print plate and send it to printer
2. The window which you see the options like bed leveling, timelapse
3. Pressing send to printer it flicks fast till it starts the countdown to forward you to Device tab and see the camera

### Actual results

The printing started normal

### Expected results

It should not flick

### Project file & Debug log uploads

log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NordicRavn, 2024-01-06:
[image]

Padawg, 2024-01-07:
I also have this happening.

erebus04, 2024-01-07:
I have the same problem

closed 2024-04-09 by NordicRavn

liciana24, 2024-07-30:
@NordicRavn @SoftFever i dont see a linked pr but still have this happen on 2.1.1