OrcaSlicer/OrcaSlicer #3515: Auto-arange moves part off plate to bottom left of screen.

TexZeTech, 2024-01-06:
### OrcaSlicer Version

1.8.1, 1.9.0

### OS version

Manjaro

### Additional system information

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.113.0
Qt Version: 5.15.11
Kernel Version: 6.1.69-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

### Printer

Custom profile. #Customized Robo3dR1+

### How to reproduce

1. have multi-build plates.
2. place two objects on build plates. one on a separate build plate and the other on the last.
3. Click the arrange object on the build plate icon on the individual plates.
4. move the 2nd object onto the last built plate.
5. click the arrange object button.
6. ?????
7. the 2nd part moves off the build plate into the bottom left.separate

### Actual results

[image]
[image]

### Expected results

To have the parts auto-arranged on the same plate.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13850855/log.zip)
[140mm_extrusion_drill_guide.zip](https://github.com/SoftFever/OrcaSlicer/files/13850858/140mm_extrusion_drill_guide.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TexZeTech, 2024-01-06:
I also have multiple materials (mainly for color coding) but I don't think this will be an actual issue in reproduction.
I also have not tested this on windows.

TexZeTech, 2024-04-09:
I will verify this in v2.0.0/dev build and post results.

closed 2024-07-16 by github-actions[bot]

chuckwagoncomputing, 2026-10-15:
This seems to happen on Kobra 3 because the because the "Excluded bed area" is a 2mm ring around the edge. You can temporarily remove that setting, then reset it later.