OrcaSlicer/OrcaSlicer #3516: Segmentation fault trying to launch

goombah88, 2024-01-06:
### OrcaSlicer Version

1.8.1, 1.9.0

### OS version

OS: fresh install of Rhino Linux (ubuntu-based)

### Printer

Bambu P1S

### How to reproduce

Double click to launch the AppImage file or run it in the terminal.

### Actual results

I get a black square in the middle of the screen until the segmentation fault and then it disappears.

user@computer:~$ /home/user/Desktop/OrcaSlicer_Linux_V1.9.0.AppImage 
[2024-01-06 14:22:48.082374] [0x00007573442fcc80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Gtk-Message: 14:22:48.211: Failed to load module "colorreload-gtk-module"

(orca-slicer:14365): dbind-WARNING **: 14:22:48.212: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
loading network module at /home/goombah/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x584e8f66f100

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.373: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.373: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.373: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.373: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.374: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.374: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.374: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.374: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.374: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.374: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.375: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.375: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.375: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.375: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.376: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.376: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.376: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.376: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.376: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.376: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.377: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.377: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:14365): GLib-GObject-CRITICAL **: 14:22:52.377: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.377: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Could not determine the accessibility bus address

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.432: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.434: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.437: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.438: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.438: gtk_widget_set_size_request: assertion 'width >= -1' failed
Gtk-Message: 14:22:52.453: Failed to load module "colorreload-gtk-module"

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.500: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.500: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.506: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.506: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.508: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.508: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.511: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.511: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.513: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.513: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.516: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.516: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.519: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.519: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.522: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:14365): Gtk-CRITICAL **: 14:22:52.522: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault

### Expected results

App should open.

### Project file & Debug log uploads

[log.tar.gz](https://github.com/SoftFever/OrcaSlicer/files/13851012/log.tar.gz)

### Checklist of files to include

- [X] Log file
- [ ] Project file

goombah88, 2024-01-06:
Other AppImages work fine - PrusaSlicer, kdenlive, etc.

mtczekajlo, 2024-01-15:
Same problem here for both 1.8.1 and 1.9.0.

Tested natively on Enderavour OS (Arch-based).
Tested in docker on both Debian 11, Ubuntu 23.10 and Fedora 39 with `--appimage-extract-and-run` option (had run `sudo ./BuildLinux.sh -u` before).

Debian 11

```bash
debian@zen:/work$ ./OrcaSlicer_Linux_V1.9.0.AppImage --appimage-extract-and-run
[2024-01-15 21:00:04.786497] [0x0000737bdd9efc80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[2024-01-15 21:00:04.796403] [0x0000737bdd9efc80] [error]   boost::filesystem::create_directory: No such file or directory [system:2]: "/home/user/.config/OrcaSlicer"

(process:7534): GLib-GObject-WARNING **: 21:00:04.796: invalid (NULL) pointer instance

(process:7534): GLib-GObject-CRITICAL **: 21:00:04.796: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

Ubuntu 23.10 in-docker

```bash
ubuntu@zen:/work$ ./OrcaSlicer_Linux_V1.9.0.AppImage --appimage-extract-and-run
[2024-01-15 20:58:21.498662] [0x00007ac78f5b5c80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[2024-01-15 20:58:21.509709] [0x00007ac78f5b5c80] [error]   boost::filesystem::create_directory: No such file or directory [system:2]: "/home/ubuntu/.config/OrcaSlicer"

(process:6379): GLib-GObject-CRITICAL **: 20:58:21.509: invalid (NULL) pointer instance

(process:6379): GLib-GObject-CRITICAL **: 20:58:21.509: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

Fedora 39 in-docker

```bash
[fedora@zen work]$ ./OrcaSlicer_Linux_V1.9.0.AppImage --appimage-extract-and-run
[2024-01-15 21:14:45.013537] [0x0000793050776d40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[2024-01-15 21:14:45.028039] [0x0000793050776d40] [error]   boost::filesystem::create_directory: No such file or directory [system:2]: "/home/fedora/.config/OrcaSlicer"

(process:79): GLib-GObject-CRITICAL **: 21:14:45.028: invalid (NULL) pointer instance

(process:79): GLib-GObject-CRITICAL **: 21:14:45.028: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

mtczekajlo, 2024-01-15:
I've just confirmed that **Cinnamon** is the root cause of my issue.
Both 1.8.1 and 1.9.0 work on Gnome (both X11 and Wayland) and KDE Plasma (both X11 and Wayland).

I'm about to check the Wayland variant of Cinnamon, yet it is still experimanetal AFAIR.

goombah88, 2024-01-15:
I was using KDE with Wayland when I had the issue in Rhino linux.   I have
no issues back in EndeavourOS (arch based) with XFCE.

mtczekajlo, 2024-01-15:
After clean reinstalling Cinnamon up to 6.0.4-1 the issue is gone. :)

lonyelon, 2024-02-07:
Seems like a dependency problem. It happens on NixOS too and it should not.

holly-hacker, 2024-03-01:
I'm not sure if this is related to the desktop environment, as I have the same issue on NixOS running Gnome under Wayland.

cvanelteren, 2024-04-01:
Same issue running arch

lonyelon, 2024-04-01:
> I'm not sure if this is related to the desktop environment, as I have the same issue on NixOS running Gnome under Wayland. Bambu Studio doesn't work either, though Prusa Slicer and Slic3r do work.
> 
> On nix, [this flake by ovlach](https://github.com/ovlach/nix-orca-slicer) seems to have patches specifically to fix a segfault, but it is also 5 months old at the time of writing so I don't know if it addresses the same issue.

I tried fixing the nix package using code from that flake and I had no luck.

cvanelteren, 2024-04-01:
My error logs are attached

<details> 

```
gdb orca-slicer                                                                                                                                                                           2024/04/01 06:33:58 PM
GNU gdb (GDB) 14.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from orca-slicer...
Reading symbols from /usr/lib/debug/opt/orca-slicer-unstable/bin/orca-slicer.debug...
(No debugging symbols found in /usr/lib/debug/opt/orca-slicer-unstable/bin/orca-slicer.debug)
(gdb) run
Starting program: /usr/bin/orca-slicer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[2024-04-01 18:34:09.065795] [0x00007fffe379f480] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffe20006c0 (LWP 10314)]
[New Thread 0x7fffe16006c0 (LWP 10315)]
[Thread 0x7fffe20006c0 (LWP 10314) exited]
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[New Thread 0x7fffe20006c0 (LWP 10316)]
[New Thread 0x7fffe0c006c0 (LWP 10317)]
[New Thread 0x7fffdbe006c0 (LWP 10318)]
[Detaching after vfork from child process 10319]
[Detaching after vfork from child process 10321]
[New Thread 0x7fffdb4006c0 (LWP 10323)]

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.287: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.287: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.288: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.288: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.289: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.289: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.289: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.289: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.290: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.290: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.290: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.290: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:10309): GLib-GObject-CRITICAL **: 18:34:09.291: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.291: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[... 82 lines cut ...]
(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.701: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.701: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.712: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.712: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.717: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.717: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.721: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:10309): Gtk-CRITICAL **: 18:34:09.721: gtk_widget_set_size_request: assertion 'width >= -1' failed

Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
0x000055555734c069 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()

```
</details>

cvanelteren, 2024-04-01:
Started working after installing `wxwidgets` on arch

alive-corpse, 2024-04-19:
The same issue with MX Linux 23 with X11 and i3 desktop environment. The log is the same as above.

alive-corpse, 2024-05-05:
I've found solving here: https://github.com/SoftFever/OrcaSlicer/issues/3160

For me installing of gstreamer1.0-x package solved the problem completely. I'm using MX Linux 23 ahs version (it's debian based). Hope it's also will work for other people.

goombah88, 2024-05-05:
The problem went away for me when they released 2.0.0. I have had no
problems since using the AppImage of that version.

Thanks,

Chad

holly-hacker, 2024-05-05:
Seems to work again on nixos as well.

lonyelon, 2024-05-06:
Here is the patch in nixos: https://github.com/NixOS/nixpkgs/blob/fc55cdb8340a3258a1ad6f3eb8df52dac36c3e70/pkgs/applications/misc/bambu-studio/0001-not-for-upstream-CMakeLists-Link-against-webkit2gtk-.patch

This makes it work.

closed 2024-08-12 by github-actions[bot]