OrcaSlicer/OrcaSlicer #3517: Auto-orient not working correctly

wardzin, 2024-01-06:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Additional system information

CPU: 11600K
Memory: 32 GB
GPU: 3080 Ti

### Printer

Bambu Lab X1C

### How to reproduce

1. Import [bullet.zip](https://github.com/SoftFever/OrcaSlicer/files/13851310/bullet.zip) into OrcaSlicer, it should start on its side.
2. Click Auto-Orient

### Actual results

Nothing happens, the bullet remains on its side.
[image]

It's also worth noting that even if you "help" it by almost orienting it correctly, it will still fail and start orienting it in strange angles.

### Expected results

The bullet should be turned upright, as this is obviously the correct orientation.
[image]

Note that Bambu Studio does actually orient the bullet correctly, so this may be a regression.

### Project file & Debug log uploads

[bullet_project.zip](https://github.com/SoftFever/OrcaSlicer/files/13851327/bullet_project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ControllerArts, 2024-01-07:
Just adding the information that in version 1.8.1, the feature was working as expected.

**V1.8.1** - Working as expected

https://github.com/SoftFever/OrcaSlicer/assets/20363585/120ba401-d9ad-4d5b-b2b9-571dcce2c44d

**V1.9.0** - Weird and unexpected behavior
the same object from the first video

https://github.com/SoftFever/OrcaSlicer/assets/20363585/9efa38b8-c387-4806-ac62-f3ac06d1165b

default / primitive cube from Orca

https://github.com/SoftFever/OrcaSlicer/assets/20363585/a7a0b798-8b57-4fd8-9295-cbbea367c487

MyStoopidStuff, 2024-01-08:
I have observed this issue as well, along with auto-rotate.  The models in the pic are some latches for a case I'm [working on](https://www.mystoopidstuff.com/blog/finally-getting-back-to-the-gridfinity-box-for-stuff) (not yet published - but I can post the models in the next few days if it would help). I first dropped them on the plate (dropped them together, but not as a "single object with multiple parts"), and then selected "arrange" (via right the click menu) followed by "auto-rotate". The result would have been an interesting printing experience I think.
[image]

CameraRick, 2024-01-09:
It isn't working here either, but worked for me in 1.8.1.
To add, when I hit the button I get a little "Orienting..." window in the lower right corner of the screen, for exactly one frame; please mind the quality of the image, I had to make a screen recording to snatch it

[image]

brjohnso1, 2024-01-10:
I have the same issue on the Mac version (1.8). Auto rotate does nothing. It may be related to the fact that auto-arrange also does nothing on the Mac 1.8 version. Both features work on Bambu Studio.

CameraRick, 2024-01-11:
Okay, this issue becomes more weird to me. I just wanted to print some part, and suddenly the Auto-Orient works - kinda. I had to do it twice. But on my original test object, it still doesn't work. I recorded a video to show what I mean:

[](https://youtu.be/-UDOjUjDVPg)

bbbrown70, 2024-01-12:
also broken here on 1.9 with model https://www.printables.com/model/644333-spool-shelf-storage
doesnt work no matter how many times the button is clicked

closed 2024-01-30 by SoftFever

timnolte, 2025-03-14:
This still seems to be an issue with more complex models with many overhangs.

PR #3890 (2024-01-30): Fix some rotation-related issues
This fixes:
- Auto orient, fixes #3517, fixes #3632
- Fix false error "The object xxx exceeds the maximum build volume height" after rotation:
[image]