OrcaSlicer/OrcaSlicer #3520: Custom Interface Color

Hyperspeed1313, 2024-01-06:
**Is your feature request related to a problem? Please describe.**
It would be nice if OrcaSlicer allowed the user to set a customizable interface color in Preferences. I'm not a fan of the teal it's currently locked to.

Ocraftyone, 2024-01-07:
Hmm. Interesting idea..  could be done via a replacement during svg import in BitmapCache. That would possibly require a restart of orca but I don't think that would be unreasonable.

closed 2024-04-14 by github-actions[bot]

NEOLabs-software, 2025-08-06:
> > just a ps, i did change the color by editing the source code came out looking like this
> > [image]
> > it was a pain but worth it in my opinion
> 
> Mind sharing how you did it?

i did a find and replace on the entire codebase
grabbed colors with an eyedropper like xcolor, and converted the color into multiple formats like hex, rgb, rgba, and a weird rgba in float format which i think is weird.
in the future though i want to make a theming engine for orcaslicer where colors are defined on a json file and would love to contact softfever for help on where all colors are defined.

Prazgreen1, 2025-10-04:
> [@Prazgreen1](https://github.com/Prazgreen1) i have begun works on the theming engine it will be similar to cura, you will be able to change the theme to whatever you want. i will create a pr once its finished

ty