OrcaSlicer/OrcaSlicer #3523: Shorter Load Line outside of Building plate

SufficentMuffin, 2024-01-06:
Since sometimes when the object is big enough it can interfer with the load line. If it is not removed soon enough it will ruin the bottom of the print.

Describe the solution you'd like
I came up with an solution where the load line is printed outside the actual printing zone

This should do the trick:

;===== noozle load line ===============================
M975 S1
G90
M83
T1000
G1 X88 Y-3 Z0.8 F18000;Move to start position
M109 S{nozzle_temperature_initial_layer[initial_extruder]}
G1 Z0.2
G0 E2 F300
G0 X162 E5 F{outer_wall_volumetric_speed/(0.3*0.5) * 60}
G0 Y-2.5
G0 E0.2
G0 X88 E5
M400

Proof of concept:

[image]
[image]

I can edit the code myself if needed I just wanted to create a feature request first and idk if it makes sense to introduce this to the Orca Community while Bambu Studio still uses the old load line. But i created there a feature request too :)

DVSVIDEO, 2024-01-07:
On Klipper machines, there is already a solution for this: KAMP  (a macro script)
https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging

SufficentMuffin, 2024-01-08:
Okay I didn´t know that. But since BabmuLab not uses Klipper this would not work I think. Its just a suggestion for shorter and less interfering load line on X1 / P1 machines.

closed 2024-04-15 by github-actions[bot]