OrcaSlicer/OrcaSlicer #3525: Enable Lidar for K1 Max. Creality allows use on Firmware 1.3.3.3

bhast2, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
Lidar feature on the K1 Max. It can be used with the new firmware 1.3.3.3
Here is the change log from Creality : LiDAR pressure advance re-worked (Only K1 Max), compatible with Creality Print, Prusa, Orca slicers

**Which printers will be beneficial to this feature**
K1 Max

**Describe the solution you'd like**
Enable Lidar feature 

**Describe alternatives you've considered**
Use a different slicer

**Additional context**
https://[raw.githubusercontent.com/Guilouz/Creality-K1-and-K1-Max/main/Changelogs/Changelog_1.3.3.3.txt](https://raw.githubusercontent.com/Guilouz/Creality-K1-and-K1-Max/main/Changelogs/Changelog_1.3.3.3.txt)

Ocraftyone, 2024-01-07:
I am a bit confused on what change needs to be made to the slicer. Is there a command that should be added to the profile or more that needs to be done?

Myklt8, 2024-01-07:
Yes, in Creality print you check a box to run the LIDAR calibration. Right now you have to download the file then go to the printer touch screen and start the print from there with calibration checked

bistory, 2024-01-08:
> Yes, in Creality print you check a box to run the LIDAR calibration. Right now you have to download the file then go to the printer touch screen and start the print from there with calibration checked

Unfortunately, Creality didn't share any information about what command has to be shared before printing... So at the moment we have to deal with what we have (send to printer then print from screen if you need PA automatic calibration).

teamfighter, 2024-03-22:
Subscribed

closed 2024-06-28 by github-actions[bot]

kanishk619, 2024-06-28:
Since I wasn't able to find anything regarding this, I did my own digging and found that you can connect to port 9999 with websocket and trigger print with AI support. Upload the file to printer and then trigger using below command

`echo '{"method":"set","params":{"opGcodeFile":"printprt:/usr/data/printer_data/gcodes/filename_to_be_printed.gcode","enableSelfTest":1,"aiPausePrint":1,"aiDetection":1,"aiFirstFloor":1,"aiSw":1}}' | ./websocat.x86_64-unknown-linux-musl ws://printer.ip.address:9999`