OrcaSlicer/OrcaSlicer #3528: Implement tab system similar to internet browsers

d-portero, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
Having multiple printers to slice for leads me to have many instances of Orca open on my computer. This leads to poorer performance because of all of the open programs. 

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
I would love if a system of tabs like a browser could be implemented. each tab could have a separate project open in it, or profiles for specific printers. This would reduce the number of instances of Orca that would need to be open at one time

Thanks!

mneuhaus, 2024-04-04:
Just throwing my 50 cents in here and showing some support, i would also love to have something like this built in, because currently i'm constantly searching for the "right" OrcaSlicer Window, all of them look the same, but i got one window per Project and/or Printer open which gets messy.

So having a Single Window with either something like a browser Tab Bar above the "Prepare, Preview, Device, Project" Tab bar, or another "sidebar" to the left of the settings Sidebar with all Open Projects would be awesome!

d-portero, 2024-07-04:
Comment to keep open

closed 2024-10-10 by github-actions[bot]