OrcaSlicer/OrcaSlicer #3530: Multi bed Temp issue

bhast2, 2024-01-07:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Additional system information

Amd Ryzen 5800X, 32G of ram. MSI Geforce RTX 2080 Gaming TRio

### Printer

K1Max

### How to reproduce

Toggle to use Multi Bed types and pick Smooth PEI Plate, Slice model and send to printer it will heat the bed up to 100 degrees instead of what is in the setting for bed temperature

### Actual results

The bed will heat up to 100 degrees instead of 60 degrees

### Expected results

60 degrees on the bed temp

### Project file & Debug log uploads

[croc boot.zip](https://github.com/SoftFever/OrcaSlicer/files/13852348/croc.boot.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bhast2, 2024-01-07:
I want to add to this if I uncheck Activate temperature control in the filament the issue goes away

DVSVIDEO, 2024-01-07:
For each material type, the specific temperatures for the table type are displayed.
Have you checked what is set there for the smooth surface PEI sheet?

bhast2, 2024-01-07:
> For each material type, the specific temperatures for the sheet type are displayed. Have you checked what is set there for the smooth surface PEI sheet?

Yes and they say 35 degrees on the smooth. but it still goes to 100

vgdh, 2024-04-06:
Slicer doesn't change bed temperature for upper layers  https://github.com/SoftFever/OrcaSlicer/issues/4676

Sharkooon42, 2024-04-07:
I have corrected my old post again for better understanding because it is not the same...

closed 2024-06-23 by bhast2