OrcaSlicer/OrcaSlicer #3533: Choosing a different Bambu printer type erases manual paint colors for multicolor prints

DKCP22, 2024-01-07:
### OrcaSlicer Version

1.9.0 happened on previous versions also

### OS version

Windows 10

### Additional system information

No performance issues

### Printer

(6)P1P's, a P1S, and a X1C happens on all 8 of them

### How to reproduce

Open any *.3mf file on a any P1P, P1S, or X1C and then change printer type to a different one. All color painting is gone on the model and all AMS filaments are reset to 1 color. Thus, having to paint the model again for the printer selection just because of switching printer types.

### Actual results

Have to re-paint the model every time

### Expected results

Changing the printer types should not have anything to do with the colors painted on a *.3mf file. It doesn't work like that in Bambu Studio.

### Project file & Debug log uploads

Happens on all *.3mf files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2024-04-14 by github-actions[bot]

DKCP22, 2024-04-14:
It still happens you lose all your filament settings when switching printer
types

auzystephens, 2024-06-27:
Can also duplicate this issue, it got worse after the v2.1.1 update, now it removes the colors and painting entirely when before it only swapped the colors around. I'm on Mac Mini M2. Let me know if I can provide additional info.