OrcaSlicer/OrcaSlicer #3534: After installation of 1.9 the device Tab appears empty

The-xyph, 2024-01-07:
### OrcaSlicer Version

V1.9.0 Official

### OS version

Windows 11

### Additional system information

13th Gen Intel(R) Core(TM) i9-13900K, 3000 Mhz, 24 Core(s), 32 Logical Processor(s)
Installed Physical Memory (RAM)	64.0 GB

### Printer

Neptune 4 PRO

### How to reproduce

- Install Orca Slicer V1.9.0
- Observe the Device Tab is white, no content or web interface link, no setup video.

### Actual results

[image]

### Expected results

The device tab should show the printer's web interface

### Project file & Debug log uploads

There are no logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

iambucketdotcom, 2024-01-09:
Issue exists on MacOS Sonoma 14.2.1 as well.

The-xyph, 2024-01-09:
So.. it actually appears that, though some settings were correctly transferred (filament/printer profiles). The connection settings were not, the Device tab did not reset to the default tutorial video/page however.

 I just needed to use the 'Connect' button and re-renter my printer's details and it works as expected.

In light of this, for QoL, I'd recommend an explicit reset of the device component to defaults OR some sort of retention of the IP/Connection settings across installs.

A further note would be to simply make the Connection icon Red and or with a cross through the icon to indicate no connection, as the current UI appears to show a green icon regardless of it's state.

Thanks so much for all your amazing work, hope this helps!

mattsfalco, 2024-01-22:
[image]

I had to reset my connection settings to get this to work. The small icon to click in the screenshot.

vistvanv, 2024-02-24:
> <img alt="Screenshot 2024-01-22 at 10 40 33" width="450" src="https://private-user-images.githubusercontent.com/5975457/298533502-737b8a6d-727b-4c1f-97ef-cca691c9448a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg4MDU4NzksIm5iZiI6MTcwODgwNTU3OSwicGF0aCI6Ii81OTc1NDU3LzI5ODUzMzUwMi03MzdiOGE2ZC03MjdiLTRjMWYtOTdlZi1jY2E2OTFjOTQ0OGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjRUMjAxMjU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTBmMzZiZDdkZDIxMDAzYmQ3YjU5YWM3N2FlZDUwYTE1ZTU5YjllMzE4OGZjOTFhMWU5ZDdiNDIxMjcyNTc5OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.cFa-B3YITVCBZMzX-8lEDwp9ol--J1sNgvkGhqnzb9M">
> I had to reset my connection settings to get this to work. The small icon to click in the screenshot.

thanks for that, i have been searching all over on how to reset my device tab

closed 2024-06-01 by github-actions[bot]