OrcaSlicer/OrcaSlicer #3535: Calibration Tab manual pressure advance errors

FlywheelMist, 2024-01-07:
### OrcaSlicer Version

1.9

### OS version

Windows 10

### Additional system information

Ryzen 9 5950x
Geforce RTX 3080
32gb ddr4
2tb NVME SSD

### Printer

Bambu A1

### How to reproduce

1. Go to "Calibration" to the right of "Project" and "Device"
2. Go to "Pressure Advance"
3. Go to "Manual" (at the bottom)
4. Without changing a thing, I get an issue with the ranges of input values.
5. Settings are: -0.4mm nozzle -Textured PEI Plate -Bambu PETG Basic (gold) -Line -0 -0.5 -0.005
6. Different error if you enter the values stated on the first popup (start=0.0 End=0.3 Value Step=0.001)

### Actual results

Popup Pictured for default settings, 

[image]
[image]

and for recommended settings from that popup,

[image]
[image]

### Expected results

It works? not sure as I haven't gotten that far

### Project file & Debug log uploads

[debug_Sun_Jan_07_13_39_45_12400.log.0.txt](https://github.com/SoftFever/OrcaSlicer/files/13853699/debug_Sun_Jan_07_13_39_45_12400.log.0.txt)
[Project File.zip](https://github.com/SoftFever/OrcaSlicer/files/13853701/Project.File.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

jlo1975, 2024-01-07:
with k from 0 to 0.3 using a step of 0.001 you will print 300 samples
not sure that would be realistic to do that and/or usefull

try with a step of 0.01 for 30 sample (that would be still a lot )

jlo1975, 2024-01-07:
with k from 0 to 0.3 using a step of 0.001 you will print 300 samples
not sure that would be realistic to do that and/or usefull

try with a step of 0.01 for 30 sample (that would be still a lot )

here a test with step of 0.005  for k=0->0.3
I'm already almost full plate

[image]

closed 2024-04-14 by github-actions[bot]