OrcaSlicer/OrcaSlicer #3536: WIFI to Flashforge M5 Pro

Millemus, 2024-01-07:
I have a Flashforge M5 Pro With Wifi
is it possible to make it so that I can use wifi without having to create filerbe for a usb key

Discorez, 2024-02-18:
In version 2.0 of the program there is a profile for this printer and downloading via WiFI works.
I found it on YouTube using this link - https://www.youtube.com/watch?v=2QhjS5pAz_8

Millemus, 2024-02-21:
it works 100%, on our m5 pro 1000 thanks for the help.

YahWhoMtnDew, 2024-03-08:
Yeah, I got my hopes WAY up when I came across that YT video, too.  Unfortunately, when I tried 2.0 dev AppImage the only new printer type in the pull-down list was something like "obico," not REMOTELY like "Flashforge" as the video demonstrates!  HUGELY disappointing, to say the least!  I have to wonder at this point if it would even be worth my trying to finger out how to build my own (maybe sketchy on this wimpy Chromebook) - if it's not there either!?!?!?!?

jtenniswood, 2024-03-08:
There is a test build that I'm using and works perfectly, I don't understand why this isn't in the nightly builds?

Millemus, 2024-03-08:
I totally agree with you

YahWhoMtnDew, 2024-03-08:
[image]

Here's a shot of what I see - clearly no Flashforge offered there!  Given that I'm working with 2.0 as you were, it's a total mystery to me why I don't have any joy.  It's so UNFAIR!!!  Waaaaaaaaaaaaaaaaaaaaaaaaaaa

jtenniswood, 2024-03-08:
Yes it doesn't show up for me. Only works on that earlier build. Still
better than copying to a usb key tho!

YahWhoMtnDew, 2024-03-08:
Well, that's precisely what I'm after: TIRED of the "sneaker-net" thing! So
I'm STILL wondering how I can lay my hands on a version LIKE YOURS!... What
OS are you running, BTW?

Millemus, 2024-03-10:
https://www.youtube.com/watch?v=2QhjS5pAz_8&ab_channel=RCSchim

watch this youtuber. and it works

[image]

YahWhoMtnDew, 2024-03-10:
> https://www.youtube.com/watch?v=2QhjS5pAz_8&ab_channel=RCSchim
>
> watch this youtuber. and it works
>
> Skaermbillede.2024-03-10.195219.png (view on web)

Well, that's precisely the video that had me running to snag v.2.0 beta for
myself, with HIGH HOPES!  But when I did so, there was NO SUCH OFFERING!  I
posted a pic of the choices I saw, so your similar pic showing something
different begs the question: WHY are we seeing DIFFERENT THINGS in what is
supposed to be the same numbered version????????  I don't see Obico in
yours and I don't see Flashforge in mine!

jtenniswood, 2024-03-10:
I just don't think it's in the Mac version yet.

YahWhoMtnDew, 2024-03-10:
Well, now, that DOES bring up a non-trivial detail: are you running Windoze or linux?  I'm certainly NOT on a Muk, as I am seriously allergic to Apples, but I'm going to have to revisit that video to see what OS it seems to be working on!  I didn't even THINK there would be different features based on which OS!?!?!?  Silly me?

YahWhoMtnDew, 2024-03-10:
Yup - it's Windoze that it's working on.  Not on Mac - couldn't care LESS - but SADLY, NOT on Linux, either!  Dunno if my wimpy little Chromebook has the capacity to try to build my own, but if it doesn't show up pretty soon in the prebuilt offerings, I may have to try to finger out how to do it!  Just waiting at someone else's mercy is PAINFUL!

YahWhoMtnDew, 2024-03-10:
Just D/Led the latest to check again: STILL NO JOY.  Same as before: Obico but no Flashforge.

How does one push a request like this forward - is this thread at all likely to come to the attention of any of the coders?

Millemus, 2024-03-10:
I run windows 11
version 23H2 build 22631.3235
[image]

- as of today (6.Feb.2024) this is the newest beta version to contain the "Flashforge" Printertype - it will for sure be in the main Orca release soon.
[https://github.com/iherbak/OrcaSlicer...](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbXlkNnU0RzliX3R3d3llUUVPbGJjakVQOTVHUXxBQ3Jtc0ttSFI1RVI4NDlMREVfOF9fb2NFSkkzd1RyZF9BU2h2dzE1Z2h6SWV6N2k1eTFpTXNqRVZCVVg0Nl9VLXNyZFhsZW96UTlKMlJWLUhKWVlEY09mZG94dGZBZ3ZkNFlQMEtWby10TWJsWEhNUEJybTYxdw&q=https%3A%2F%2Fgithub.com%2Fiherbak%2FOrcaSlicer%2Freleases%2Fdownload%2FV2%2FOrcaSlicer_A0vy.zip&v=2QhjS5pAz_8)     (download the above ZIP and extract for Windows PC)

There's also a mac version:
[https://github.com/iherbak/OrcaSlicer...](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqa056VTlhbDlKNXYzRmtJM0Z4ZDYyWjRCdFRjZ3xBQ3Jtc0tuOUxqakRZWXBCZG8xZDdqd212dzRVbGFXZnRnSjZNek1DNTVLczZBRFpyVURtcE1kNUVYNlRCalVwSHprZC1EVTN4UG9FeDdWRlhKQzBCTXNHMnVSVFAyUUV4N0FfYkp2THVCbkRKRl9mbVhfVXhGdw&q=https%3A%2F%2Fgithub.com%2Fiherbak%2FOrcaSlicer%2Freleases%3Ffbclid%3DIwAR1KyE1_eQlgBOxUEfNy9J7aCZ3DTqRWx4vlb2DPfM6ms4W4aEJfTKZEagA&v=2QhjS5pAz_8)

Millemus, 2024-03-10:
[image]

from inside my Flashforge M5 Pro

Millemus, 2024-03-10:
192.168.0.225
http://192.168.0.225:8080/stream_simple.html

Millemus, 2024-03-10:
[image]
[image]

YahWhoMtnDew, 2024-06-10:
Or - it's been resolved - in the latest release?

closed 2024-09-16 by github-actions[bot]