OrcaSlicer/OrcaSlicer #3538: Slicer non-deterministic results

itay-grudev, 2024-01-07:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 23.10

### Additional system information

Ryzen 9 7950X3D
GTX 3070

### Printer

Creality K1 Maz

### How to reproduce

I have a component that has a small hole that goes through the component. Sometimes when I slice, **with exactly the same settings**, I get this:

[image]

Most of the times, I don't get anything:

[image]

And this is the part that really puzzles, because I would really expect a software like this to be deterministic with results that are highly reproducible.

### Actual results

Significant variations between slices.

### Expected results

I expect that between slices with exactly the same settings, the results are the same.

### Project file & Debug log uploads

[Orca Project + Logs.zip](https://github.com/SoftFever/OrcaSlicer/files/13853803/Orca.Project.%2B.Logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DVSVIDEO, 2024-01-07:
I tried it.
This is a faulty model.
Before slicing, right-click on the object and select the repair object menu item.
You will get a message that the object is repaired.
Then it will slice perfectly and you will always get the same

Since this is not a real error, I suggest you clear the error sheet with a short explanation at the bottom (there is enough error reporting).

itay-grudev, 2024-01-07:
Even if the model is faulty, why isn't the result consistent? Shouldn't the same input should lead to the same output.

I suppose this is especially relevant on Linux, where the Fix Model feature does not work.

DVSVIDEO, 2024-01-07:
Wrong model, wrong result.
Accept it.  Move on...

DVSVIDEO, 2024-01-11:
> Even if the model is faulty, why isn't the result consistent? Shouldn't the same input result in the same output.
> 
> I suppose this is especially relevant on Linux, where the Fix Model feature is not supported.

If you start off with a flat tyre, you will always end up somewhere else again and again. Sometimes in one ditch, sometimes in another.
Because the wheel is defective, it introduces an error into the system, which makes it unpredictable.
If you start with a flawless wheel, you'll always end up in the same place.
I hope you understand the analogy.
A faulty model can lead to faulty calculations, which can vary.
Solution: - Use a good model
                 - Correct the model with Orcaslicer
                 - Correct model with 3rd party program (meshmixer)

itay-grudev, 2024-01-15:
Here is the thing, one would expect that given the same input file, the slicer algorithm does the same thing and produces the same G-Code every single time. This is referred to as a [deterministic algorithm](https://en.wikipedia.org/wiki/Deterministic_algorithm) in Computer Science.

But in reality that doesn't seem to be the case in the slicer. There is an element of random (or another decision-making source) that branches the algorithm in a different way and changes the result between slices.

Yes, the file has errors, but it is likely indicative of a deeper issue in the slicing engine.

DVSVIDEO, 2024-01-18:
I disagree with you.
When the input file is OK, the slicer handles it accurately, I always get the same result.
But if the model is wrong, the floating point calculations can be wrong. These errors act like a random number generator. They may give a different result after each run.
I think this would only need to be fixed if it could happen at any time.
After I corrected the model you posted, each run gave the same correct result.
Conclusion: the model needs fixing and not the slicer.

dstulken, 2024-04-18:
> When the input file is OK, the slicer handles it accurately, I always get the same result.
> But if the model is wrong, the floating point calculations can be wrong. These errors act like a random number generator. They may give a different result after each run.

I don't agree.

A model should slice the same way each time.
If there is a fault in the model, some logic in the parsing has to decide how to handle that fault.
This fault handling should be deterministic - and since the model hasn't changed between runs, a faulty or corrupt model should still be processed the same way each time.  There should not be any floating point values "acting as random number generators", because those floating point values should always be the same each time.

The fact that you get the correct result when the input file is not corrupt doesn't prove that the model is the issue here - you have simply removed the test case that was exposing the nondeterministic behavior in the slicing logic.

itay-grudev, 2024-07-18:
Ping.

itay-grudev, 2024-10-18:
Ping.

jaydenseric, 2024-12-05:
My models are not faulty (as far as I know), yet without having changed any settings, each slice the model filament estimation changes by around 0.5g. Sometimes it's the same several times in a row, sometimes it comes up with a number I don't see again. This smell like a bug, because the slicing and the estimations should be deterministic. Perhaps it's related to the bug revealed by the circumstances described in this issue?

I wondered if there is deliberate randomness in the slicing process, but as far as I know I'm not using a setting such as randomised seam placement. Is there a list of non-deterministic settings and values to check? If randomness is deliberately present in every slice, can that approach be revised to be deterministic? If not, can a RNG seed be configured, for deterministic randomness across slices?

For context, I have a very strict weight target for the items being printed and I am trial and error tweaking the space infill density setting to the exact decimal value that achieves the target model filament estimate. This is challenging when the slicer comes up with random results for the same values.

closed 2025-03-13 by github-actions[bot]