OrcaSlicer/OrcaSlicer #3540: First layer flowrate

jedikaal, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
My bed is not the most even ones out there. While the adhesion for smaller prints are great. (Mesh density) Larger prints are suffering. 
In klipper, while printer I can solve this is issue manually for the first layer by adding 8% flow rate. 

**Which printers will be beneficial to this feature**
Every printer with a slithly warped bed. 

**Describe the solution you'd like**
An extra option in the printer setting or the filament setting. So that I can define an % or a raw value for the flowrate on the first layer.

**Describe alternatives you've considered**
Manually adding it in klipper while it is printing. This is highly manual and not really reproducible for multiple prints.

**Additional context**
-

Hopefully I'm clear in the request. I would like to support with this if needed.

DVSVIDEO, 2024-01-07:
This feature has been part of Orcaslicer for quite some time.
Why don't you take a good look.
In the "Quality" tab---->"Wall and surfaces" section----->"Bottom surface flow ratio"
1 means 100%. Set it to 1.2 (120%)
By the way, there is also a "Top surface flow ratio" for the top layer.

Find out before you write...

jedikaal, 2024-01-07:
> This feature has been part of Orcaslicer for quite some time. Why don't you take a good look. In the "Quality" tab---->"Wall and surfaces" section----->"Bottom surface flow ratio" 1 means 100%. Set [...]

Woow! I had never found out without your message. Thank you for helping. 
I've searched for it but for some reason I did not interpret the words from that function as the function that I wanted. 

Your last line seems a salty but that could be miscommunication. Thank you again.

closed 2024-01-07 by jedikaal

stueble, 2024-01-26:
Is this feature broken in 1.9.0? I tried different settings but no change in the resulting GCode files. My assumption is that  this setting should also change the flow of the brim? This is the place where I compared the GCode files...