OrcaSlicer/OrcaSlicer #3545: Calibration/Temperature model issue on Mac

Scootworks, 2024-01-07:
### OrcaSlicer Version

1.9.0

### OS version

MacOS Sonoma

### Additional system information

MacBook Air M2, 2022

### Printer

no printer selected

### How to reproduce

1. New Project
2. Calibration
3. Temperature -> OK
4. Error

### Actual results

Object name: temperature_tower.st/
Size: 44.4877 x 10 x 50 mm
Volume: 11491.9 mm3
Triangles: 13776
Error: 14 non-manifold edges.
Tips:
"Fix Model" feature is currently only on Windows.
Please repair the model on Orca Slicer(windows) or
CAD softwares.

### Expected results

[image]

### Project file & Debug log uploads

[Nozzle temperature test.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13854367/Nozzle.temperature.test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Scootworks, 2024-01-07:
same happens for the speed test structure, reaction test too

StayPirate, 2024-01-27:
same on Linux and orcaslicer v1.9.0.

jel111, 2024-01-28:
Same happening to me on MBP M1 Sonoma. I printed and it worked but I am wondering if it is ok a calibration print. It also says it will only fix on Windows machine.

dekay, 2024-02-02:
Looks like a duplicate of #3349 and #1842, the latter going back to Aug 2023.

Bambuslicer also cannot repair these problems on Mac / Linux because they depend on the Windows SDK for that.

https://github.com/bambulab/BambuStudio/issues/1585#issuecomment-1585367329

poudenes, 2024-02-24:
Here same issue with MacOS 13.6.4 and OctaSlicer 1.9.1

Sjullep, 2024-05-20:
Same problems here Orcaslicer 2.0, MacOs Sonoma 14.2.1

ajaytanna, 2024-06-18:
Still the same on both MacOS Sonoma 14.5 and OrcaSlicer 2.1.0 rc as well as Windows. Temperature tower has non-manifold edges and needs repair. Repaired on windows, have to export g-code  because the saved .3mf file loses the temperature-layer change data.

eburlaka, 2024-07-06:
Still the same issue on MacOS Sonoma 14.4.1 and Orca Slicer 2.1.1

guipassos, 2024-07-27:
I've the same problem in MacOS 14.4.1

WayneManion, 2024-07-31:
I am experiencing this on latest release of Orca Slicer on Arch Linux. Is it not possible to fix the model used in the calibration so that it doesn't throw this error?

neich03, 2024-10-07:
same here. Orca Slicer 2.1.1 and Sonoma 14.6.1.
I upgraded from big sur to have O.S working and now I have to use windows 😭. Mac sucks...

67RS427, 2024-10-28:
I'm still seeing this bug as well on the latest Orca v2.2.0 on MacOS

melbiz447, 2024-11-12:
Here too, posted picture somewhere else (manifold linux), I'm on MacBook Air 13 inch 2019 intel, Mac OS 14.7.1, was on v2.1.1 and had the same issue. Now on v2.2.0 and still there. I don't have access to a PC.
[image]

wisess, 2024-12-08:
Same issue on Linux Mint slicer ver. 2.2.0

emutier, 2025-01-07:
still there in 2.3.0 on linux mint. @Scootworks: please change the issue name to "mac and linux mint"

jkga, 2025-01-24:
The same issue occurs on a MacBook M3 Pro running macOS Sequoia 15.2 with OrcaSlicer version 2.2.0

thkrmr, 2025-02-05:
M3, macOS 15.2, 2.2.0 - same issue

StephanCassel, 2025-04-13:
Same on M4 Pro. macOS 15.4 Orca 2.3.0

adastra1826, 2025-04-15:
Has anyone found a way to fix this error? I've tried online tools but they always seem to mess the model up.

adastra1826, 2025-04-15:
Using Blender, I was able to fix a file. Blender has a 3D Tools extension (which you may need to download and/or enable), which has a "Make Manifold" button. It was really that simple. Use ChatGPT for specific instructions but that's all I needed to do.

Sjullep, 2025-07-15:
Stiil an issue