OrcaSlicer/OrcaSlicer #3546: Print preview data broken on Ender 3 V3 SE

dmcpacks, 2024-01-07:
### OrcaSlicer Version

1.9.0

### OS version

MacOS 14.1.2

### Printer

Ender 3 V3 SE

### How to reproduce

1. Add an STL.
2. Slice STL
3. Export STL to sd card
4. plug sd card into printer
5. Open gcode file

### Actual results

Data for print will appear as 0's while it should appear in time or measurements. Creality Print 4.3.9 has fixed this issue, could use that as a base.

### Expected results

The data should contain actual values, images for both cases have been attached.

Note 1: gcode for both cases and images have been attached with corresponding names for the slicer that sliced.
Note 2: as far as I've seen the print preview image is an issue with printer software itself not slicer, it will always be the default creality man.

Creality Print 4.3.9
[image]
Orca Slicer 1.9.0
[image]

### Project file & Debug log uploads

Zip with gcode for both Creality Print slice and Orca Slicer slice, also includes project file for Orca Slicer.
[Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/13854572/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bistory, 2024-01-08:
I have made some tests and it seems the screen only reads those data generated by cura engine at the very begining of each file : 

```
;FLAVOR:Marlin
;TIME:3508.43
;Filament used:4.85155m
;Layer height:0.2
```

Unfortunately, there is no easy way to add those informations before HEADER_BLOCK_START in Orca...

dmcpacks, 2024-01-08:
As far as I can tell there is no way to modify it within orca slicer which is why I submitted it as a bug, a potential fix could be to allow for header data to be added by the user in the gcode options? However I'm not sure if theres a way or a variable to add the layer height, used filament, and print time.

dmcpacks, 2024-01-18:
I made a post processing script that adds the data to the screen. (filament use is provided in grams not meters btw).

Called from orca with "python3.10 /Users/user/Desktop/pyprojects/gcode/test.py;" MacOS.
All the script does is read the gcode file to find the data and write it at the top.
[test.py.zip](https://github.com/SoftFever/OrcaSlicer/files/13971787/test.py.zip)

Ubiquitous0, 2024-03-04:
Thanks @dmcpacks. Worked great! Just had to change a couple lines.

closed 2024-06-10 by github-actions[bot]

beyondsoft978, 2024-06-24:
The same situation applies to Windows. How can I solve it?

Ubiquitous0, 2024-06-24:
> The same situation applies to Windows. How can I solve it?

You can run python in Windows. Same steps apply. I'm running Windows myself.

beyondsoft978, 2024-07-11:
> > The same situation applies to Windows. How can I solve it?
> 
> You can run python in Windows. Same steps apply. I'm running Windows myself.

thanks,is good work!!!

kingKzK, 2024-07-12:
> I made a post processing script that adds the data to the screen. (filament use is provided in grams not meters btw).
>
> Called from orca with "python3.10 [...]

Can you explain me a litte more about how the script works?