OrcaSlicer/OrcaSlicer #3549: Orca 1.9.0 on Bambu.  Slicer shows gcode after slicing

boethos3, 2024-01-07:
### OrcaSlicer Version

1.9.0

### OS version

Win 10 pro

### Additional system information

Intel Xeon E3-1220 v2 3.10 ghz, 4 cores    32gb ram

### Printer

Bambu X1C

### How to reproduce

Slice model, 

### Actual results

see gcode

### Expected results

don't normally see gcode here.

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13854872/log.zip)
[Project file 3mf Error screen.zip](https://github.com/SoftFever/OrcaSlicer/files/13854873/Project.file.3mf.Error.screen.zip)

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13854874/log.zip)
[Project file 3mf Error screen.zip](https://github.com/SoftFever/OrcaSlicer/files/13854875/Project.file.3mf.Error.screen.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

giveen, 2024-01-08:
Its a new feature.
Press "C", it will go away

drackp2m, 2024-02-26:
> Its a new feature. Press "C", it will go away

my mother!!! You don't know how long I've been moving the camera so I could put the pieces in a lake where that thing wouldn't get in the way, and it was as easy as pressing a key. u_u

Thank you very much @giveen!! Really.

closed 2024-06-03 by github-actions[bot]