OrcaSlicer/OrcaSlicer #3552: Temps are set to 0C When using start print macro

Awesome4all, 2024-01-08:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Additional system information

CPU:Ryzen 5 5600x
RAM:32GB DDR4 3200Mhz
GPU:ROG STRIX 3060ti

### Printer

Ender 3 v2 Neo running Klipper

### How to reproduce

1. Open Orca Slicer
2. Click Create New Project 
3. Select the the Creality ender 3 v2  system profile
4. save as new profile 
5. edit start gcode to `START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]`
6. save profile 
7. add any new model
8. press slice plate

### Actual results

After slicing the plate under color schemes, change to temperature. it shows 0c.

### Expected results

After slicing the plate under color schemes, change to temperature. it shows 220c.

### Project file & Debug log uploads

[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/13855649/logs.zip)
[3mffiles.zip](https://github.com/SoftFever/OrcaSlicer/files/13855656/3mffiles.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

HakunMatat4, 2024-01-08:
That looks wrong, I have my Voron set like so:

`PRINT_START HOTEND={first_layer_temperature[initial_extruder]} BED=[first_layer_bed_temperature]`

HakunMatat4, 2024-01-08:
Also, even so your variables might be correct, the single major problem from Ender folks coming from Cura is that their macros are incomplete.
Meaning, `PRINT_START` does not have the nozzle and bed variable set so the macro does not know what to do with that information. 
This is more an user/printer/klippepr issue than slicer issue itself.

Awesome4all, 2024-01-08:
Nope the issue still occurs.

Awesome4all, 2024-01-08:
> Also, even so your variables might be correct, the single major problem from Ender folks coming from Cura is that their macros are incomplete. Meaning, `PRINT_START` does not have the nozzle and bed [...]

also i initially started on prusa nad then to orca and then to klipper +orca

Awesome4all, 2024-01-08:
I have both print_start and Start_print
`[gcode_macro print_start] 
gcode:
  {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
  {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(220)|float %}
  POWER_ON_PRINTER
  G4 P10000 ; wait for printer to turn on
  G90 ; use absolute coordinates
  M83 ; extruder relative mode
  M140 S{BED_TEMP} ; set final bed temp
  M104 S{EXTRUDER_TEMP} ; set temporary nozzle temp to prevent oozing during homing
  G4 S10 ; allow partial nozzle warmup
  G28 ; home all axis
  BED_MESH_CLEAR
  BED_MESH_CALIBRATE
  G1 Z50 F240
  G1 X2 Y10 F3000
  M190 S{BED_TEMP} ; wait for bed temp to stabilize
  M109 S{EXTRUDER_TEMP} ; wait for nozzle temp to stabilize
  LINE_PURGE

[gcode_macro start_print] 
gcode:
  {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
  {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(220)|float %}
  POWER_ON_PRINTER
  G4 P10000 ; wait for printer to turn on
  G90 ; use absolute coordinates
  M83 ; extruder relative mode
  M140 S{BED_TEMP} ; set final bed temp
  M104 S{EXTRUDER_TEMP} ; set temporary nozzle temp to prevent oozing during homing
  G4 S10 ; allow partial nozzle warmup
  G28 ; home all axis
  BED_MESH_CLEAR
  BED_MESH_CALIBRATE
  G1 Z50 F240
  G1 X2 Y10 F3000
  M190 S{BED_TEMP} ; wait for bed temp to stabilize
  M109 S{EXTRUDER_TEMP} ; wait for nozzle temp to stabilize
  LINE_PURGE`

Awesome4all, 2024-01-08:
either way the slicer still wont show that im printing at 240 c

Awesome4all, 2024-01-08:
[image]

bistory, 2024-01-08:
Please export and share a gcode file

Awesome4all, 2024-01-09:
I figured it out when putting the start macro in orca you have to match it with the start macro in klipper
so for this section in klipper
`{% set BED_TEMP = params.BED_TEMP|default(60)|float %} 
{% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(220)|float %}`
i originally had 
`PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single] PRINT_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} PRINT_MAX={first_layer_print_max[0]},{first_layer_print_max[1]}`
what needed to be changed was the `EXTRUDER` and `BED` section to `EXTRUDER_TEMP` and `BED_TEMP` to match the `prams.EXTRUDER_TEMP` and the `prams.BED_TEMP` section 

thank you for you help

closed 2024-01-09 by Awesome4all

HakunMatat4, 2024-01-09:
> I figured it out when putting the start macro in orca you have to match it with the start macro in klipper
> so for this section in klipper
> `{% set BED_TEMP = params.BED_TEMP|default(60)|float %}
> {% [...]

Yup, but again this isn't an OrcaSlicer problem.
Your Klipper macro was previously set with custom variables names instead if the variables commonly used by Klipper printers.
It was the first thing I mentioned, Ender folks with this same problem appears often on discord. 

[image]