OrcaSlicer/OrcaSlicer #3554: Unable to export filament preset

lin-ycv, 2024-01-08:
### OrcaSlicer Version

1.9

### OS version

Windows 11

### Printer

V400

### How to reproduce

1.Export Config
2.Filament Presets

### Actual results

Missing options to export
[image]

### Expected results

Have options like when exporting other presets
[image]

### Project file & Debug log uploads

UI design issue, not a project or a crash

### Checklist of files to include

- [X] Log file
- [ ] Project file

tlhintoq, 2024-01-10:
Confirmed on Win10x64 as well.

**Also note:**
If you have Orca on a secondary monitor, the dialog jumps to the primary after you tick a radio button.  IE: at the point it would repaint with the options appropriate for the button you ticked.
[image]

Altblechkutscher, 2024-02-25:
I've also experienced that issue.

kiesan88, 2024-02-27:
Confirming same issue (Filament presets and Filament bundle)

hannut, 2024-04-14:
Same here. Version 2.0.0 Linux AppImage

closed 2024-07-22 by github-actions[bot]

talentedmrripley, 2024-07-24:
This issue is ongoing in 2.1

timboll, 2025-05-26:
still an issue for me.