OrcaSlicer/OrcaSlicer #3555: Unable to change printing order

HakunMatat4, 2024-01-08:
### OrcaSlicer Version

Any

### OS version

Linux Mint Cinnamon

### Printer

Voron Trident

### How to reproduce

1. Add multiple objects to the plate
2. Try to have them printed the order you need

### Actual results

As much I love OrcaSlicer, this has to be worst object management ever.
We have talked about this on Discord a bunch of time but nothing was done about it.

As you are looking at it, any 3D printer I am aware of, will move the head from the left to right where is usually the purge line is printed.
With that being said, it should start printing like so, from the left to the right, from the front towards the back. 
Basic logic right?!?

Here you have the normal behaviour

[image]

A slightly change and there goes the toolhead traveling across the bed to start the print.

[image]

And again.

[image]

That data isn't helpful enough???
Sure...
What logic is this???

It is so frustrating because this is such a basic must have logic but doesn't seem to be relevant.
Nobody is talking about print by object, save that nonsense.
You have to waste minutes trying to move things around to get a somewhat decent sequence.

The object list and nothing is the same, it does not work, you cannot interact with it, not like PrusaSlicer does.
You don't have to have print by object enabled on PrusaSlicer to allow the order to be changed.

I cannot get the sequence right with a full plate like this no matter what I try.

[image]

### Expected results

Follow the common logic sequence, from the left to the right, from the front towards the back.
You avoid a 260c nozzle traveling all the way to the back to start a print which usually ends up with curling line at the start and messed up first layer.

Users with the "print by object nonsense", please, press "Alt + F4"
Thank you

### Project file & Debug log uploads

This can be simulated with anything, it does not require user log.

### Checklist of files to include

- [X] Log file
- [ ] Project file

HakunMatat4, 2024-01-08:
> I know what you are trying to achieve. I just want to point out that [prusa3d/PrusaSlicer#5175](https://github.com/prusa3d/PrusaSlicer/issues/5175) [supermerill/SuperSlicer#3033](https://github.com/supermerill/SuperSlicer/issues/3033) and things like https://forum.prusa3d.com/forum/prusaslicer/print-order/
> 
> Though I definitely agree be able to have a predicable/adjustable print sequence can be very useful.

Holy smokes, 2020?? oO

But ironically, I never encountered problem with PrusaSlicer, it allows me to change the sequence.
Sure, it is a manual process but it used to work just fine, OrcaSlicer does not even allow you to do that and you can very barely change the sequence when printing by object which again, you need to move things around and I have had many close calls because the travel was not what I had planned and avoided a toolhead crash.

Just to make sure I am not dreaming, I just tried and like always, printing by layer follows the sequence I need it to follow.
Sometimes it does have a mind of its own so I am guessing it is the bug you mentioned, but works 99% of the time
If OrcaSlicer had similar behavior with the bug included it would be so much better than it is atm, it does't work at all.

Thanks man

[image]

HakunMatat4, 2024-01-08:
> Cura behaves as: First in, last out.
> Orca (so far to me) behaves as First in , first out.
> 
> So... Seems pretty natural (to me) to import and place parts in the order you want them to print.
> If you want left to right, front to back, then import and place them in that order.

Oh believe me, I've tried that but it somewhat works now and don't work anymore onwards.
It's frustrating fighting the slicer to keep an order that makes sense.

Noisyfox, 2024-01-14:
I am working on improving the print order when it is turn off, and I just need to confirm that does the order of the object list ALWAYS work as expected WHEN "Print py object" is turned ON? Or it is also broken? I need to make things clear so I could know whether I need to fix both.

Noisyfox, 2024-01-14:
BTW, I've checked the code and it uses a nearest neighbor search in print by layer mode. It takes the original order from the object list, then try to reorder it to find out a shorter path, that's why the object list order works when there are only two objects on the plate, then it works seemingly randomly when there are more than 2.

Noisyfox, 2024-01-14:
And a bit more explaination of current code:
https://github.com/SoftFever/OrcaSlicer/blob/b7b22eb78f1193a691ee47f11b60bbcd3aaefdf4/src/libslic3r/ShortestPath.cpp#L1978-L2000

It uses the center of each object to calculate the shortest path.

Noisyfox, 2024-01-14:
My current plan is to adding a "print order" (or whatever sounds better) option, that has the following options:

- Default, which keeps the current shortest path sort
- As object list, which uses the exact order as the object list
- (tbd) Top-bottom,left-right (ie, by column)
- (tbd) Left-right, top-bottom (ie, by row)

HakunMatat4, 2024-01-15:
> My current plan is to adding a "print order" (or whatever sounds better) option, that has the following options:
> 
> * Default, which keeps the current shortest path sort
> * As object list, which uses the exact order as the object list
> * (tbd) Top-bottom,left-right (ie, by column)
> * (tbd) Left-right, top-bottom (ie, by row)
> * (tbd) maybe diagonally?
> 
> I'll first implement the first two options, then consider adding pre-defined orders. The way each object fits into rows/columns will be tricky though.

Thank you so much for looking into this, it has been so frustrating.
I am printing ASA like non-stop, by the time the 260c nozzle reach those places using the first layer speeds, you have curled oozing which often ends up as terrible first layer.

The first two options should solve all the problems, shortest path after the purge line which is usually at the very front, front corner or those using adaptive mesh can have the purge to be right next to the model.

The second option, it would be awesome to rather than just follow the object list, to allow the users to interact with it like what somewhat happens with "print by object". You might wanna start from a less risk bigger model over a small one which might not be in order so you need to move it up/down.
I understand that the logic might not be as straightforward but if Orca follows the list order, in the worst case scenario you need to conscientiously add the models following the printing sequence you wanna and we are good.

[image]

Noisyfox, 2024-01-15:
> in the worst case scenario you need to conscientiously add the models following the printing sequence you wanna and we are good.

I'll allow reordering the object list in print by layer mode so that's not a problem at all (in fact that's the first thing I did so far).

> shortest path after the purge line

Ah that's a good point... Currently the purge line is not treated as an object so its location won't be considered at all when calculating the print order. Will see if I could solve this. Worst case scenario we could use the order of the object list though.

HakunMatat4, 2024-01-15:
> I'll allow reordering the object list in print by layer mode so that's not a problem at all (in fact that's the first thing I did so far).
>
>
> Ah that's a good point... Currently the purge line is not [...]

100%, whatever you think is easier to implement will be already better than currently, it is painful. 
Thanks again for looking into this.

Noisyfox, 2024-01-15:
Please give #3685 a try and see if it works for you.

HakunMatat4, 2024-01-15:
> Please give #3685 a try and see if it works for you.

1AM now, can try tomorrow after COB 🥲
Hopefully somebody can try before that 
Thanks man 🙏

tlhintoq, 2024-01-15:
> > Please give #3685 a try and see if it works for you.
> 
> 1AM now, can try tomorrow after COB 🥲 Hopefully somebody can try before that Thanks man 🙏

0015hrs - Waiting for the actions to complete - Still has 4 spinny wheels

HakunMatat4, 2024-01-17:
> Please give #3685 a try and see if it works for you.

Unless I am doing something wrong, it is not following the object list, in fact, it is doing the opposite order, bottom/center/top
I even moved the far right one a little towards the back in case the logic is using front/back but nope.

[image]

Noisyfox, 2024-01-17:
> > Please give #3685 a try and see if it works for you.
> 
> Unless I am doing something wrong, it is not following the object list, in fact, it is doing the opposite order, bottom/center/top I even moved the far right one a little towards the back in case the logic is using front/back but nope.
> 
> ![Peek 2024-01-17 18-09](https://private-user-images.githubusercontent.com/35049512/297289952-412b407d-4a93-420f-99e1-45af8e74d1ac.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzYzMTEsIm5iZiI6MTcwNTQ3NjAxMSwicGF0aCI6Ii8zNTA0OTUxMi8yOTcyODk5NTItNDEyYjQwN2QtNGE5My00MjBmLTk5ZTEtNDVhZjhlNzRkMWFjLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDA3MjAxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkYjkxODQxMGFmY2M5YWI5MjZiMjM1Mzc4ODVhYTRkYmJkMTg3MTA0YmFmN2NkNGVlMGE4ZDFiY2ZhNTg5N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.I23KLGYDbxbaqWV17rjlYZGh1oD31pdBph-3cEy2olw) [ ![Peek 2024-01-17 18-09](https://private-user-images.githubusercontent.com/35049512/297289952-412b407d-4a93-420f-99e1-45af8e74d1ac.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzYzMTEsIm5iZiI6MTcwNTQ3NjAxMSwicGF0aCI6Ii8zNTA0OTUxMi8yOTcyODk5NTItNDEyYjQwN2QtNGE5My00MjBmLTk5ZTEtNDVhZjhlNzRkMWFjLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDA3MjAxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkYjkxODQxMGFmY2M5YWI5MjZiMjM1Mzc4ODVhYTRkYmJkMTg3MTA0YmFmN2NkNGVlMGE4ZDFiY2ZhNTg5N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.I23KLGYDbxbaqWV17rjlYZGh1oD31pdBph-3cEy2olw) ](https://private-user-images.githubusercontent.com/35049512/297289952-412b407d-4a93-420f-99e1-45af8e74d1ac.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzYzMTEsIm5iZiI6MTcwNTQ3NjAxMSwicGF0aCI6Ii8zNTA0OTUxMi8yOTcyODk5NTItNDEyYjQwN2QtNGE5My00MjBmLTk5ZTEtNDVhZjhlNzRkMWFjLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDA3MjAxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkYjkxODQxMGFmY2M5YWI5MjZiMjM1Mzc4ODVhYTRkYmJkMTg3MTA0YmFmN2NkNGVlMGE4ZDFiY2ZhNTg5N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.I23KLGYDbxbaqWV17rjlYZGh1oD31pdBph-3cEy2olw) [ ](https://private-user-images.githubusercontent.com/35049512/297289952-412b407d-4a93-420f-99e1-45af8e74d1ac.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzYzMTEsIm5iZiI6MTcwNTQ3NjAxMSwicGF0aCI6Ii8zNTA0OTUxMi8yOTcyODk5NTItNDEyYjQwN2QtNGE5My00MjBmLTk5ZTEtNDVhZjhlNzRkMWFjLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDA3MjAxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkYjkxODQxMGFmY2M5YWI5MjZiMjM1Mzc4ODVhYTRkYmJkMTg3MTA0YmFmN2NkNGVlMGE4ZDFiY2ZhNTg5N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.I23KLGYDbxbaqWV17rjlYZGh1oD31pdBph-3cEy2olw)

Could you post the project file?

HakunMatat4, 2024-01-17:
@Noisyfox , please let me know if you can open it.
Thank you

[project_3555.zip](https://github.com/SoftFever/OrcaSlicer/files/13959989/project_3555.zip)

Noisyfox, 2024-01-17:
> @Noisyfox , please let me know if you can open it. Thank you
> 
> [project_3555.zip](https://github.com/SoftFever/OrcaSlicer/files/13959989/project_3555.zip)

[image]
It seems that you did not set this option to "As object list"

HakunMatat4, 2024-01-17:
> > @Noisyfox , please let me know if you can open it. Thank you
> > 
> > [project_3555.zip](https://github.com/SoftFever/OrcaSlicer/files/13959989/project_3555.zip)
> 
> [image]
> It seems that you did not set this option to "As object list"

I think there's a misunderstanding lol
Isn't this PR changes aimed to fix the chaotic order when printing by layer??

By object isn't perfect but works, the problem is by layer, if follows no sequence at all.

Noisyfox, 2024-01-17:
> I think there's a misunderstanding lol Isn't this PR changes aimed to fix the chaotic order when printing by layer??
>
> By object isn't perfect but works, the problem is by layer, if follows no [...]

No no no, this is a separate option called "Layer order".

HakunMatat4, 2024-01-18:
> > > > @Noisyfox , please let me know if you can open it. Thank you
> > > > [project_3555.zip](https://github.com/SoftFever/OrcaSlicer/files/13959989/project_3555.zip)
> > > 
> > > 
> > > ![image](https://private-user-images.githubusercontent.com/1537155/297322354-018d3480-b752-4c0e-8698-c5df3b1d7ba3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0ODUxNjMsIm5iZiI6MTcwNTQ4NDg2MywicGF0aCI6Ii8xNTM3MTU1LzI5NzMyMjM1NC0wMThkMzQ4MC1iNzUyLTRjMGUtODY5OC1jNWRmM2IxZDdiYTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMTdUMDk0NzQzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGMzNTM2MWRhMDM0ZDlmZWQ5ZDQ2NGJkZWJjNzg0YjM5MTBiZmI5ZGE4MzE4MTkxMzA1MDJjZjFiNDUyNjdjZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.SIfxqlFeGCuQLCSZEoRcvuOzOCXAR9pfKCWzci3aleo)
> > > It seems that you did not set this option to "As object list"
> > 
> > 
> > I think there's a misunderstanding lol Isn't this PR changes aimed to fix the chaotic order when printing by layer??
> > By object isn't perfect but works, the problem is by layer, if follows no sequence at all.
> 
> No no no, this is a separate option called "Layer order", which is "the print order within each layer". Though I do think I need a better naming for this option... Any suggestion?

Of course, it has to be user doing user stuff lol
Just sent a print and it works lovely, it was a 2 print object and it does follow the list order.
Legend <3

HakunMatat4, 2024-01-18:
EDIT: "Layer order" sounds about it, perhaps add more context to its description when hovering the mouse.
Any slicer already have by layer/by object, layer order sounds just about right to me of course.

HakunMatat4, 2024-01-19:
@Noisyfox I would hug you if I could :heart: 

Hopefully @SoftFever can merge your changes then we have everything we need, a gold printing order and igianna changes ( already merged) to reduce extra non-necessary solid infill.
I am closing this ticket.

Thank you so much again for this beautiful work, and I should had built a bigger printer, 300mm is no longer enough haha

No more fights with Orca :pray: 

[image]

closed 2024-01-19 by HakunMatat4

dcchillin46, 2024-01-25:
> No no no, this is a separate option called "Layer order", which is "the print order within each layer". Though I do think I need a better naming for this option... Any suggestion?

i just happened to have a need for this and found it merged into the nightly orca build. i have about 50 objects on my plate and it seemed to get the order correct.

if youre still searching for name suggestions i would say "Intra-Layer Order" is pretty descriptive, "Layer Order" is a bit ambiguous. just my 2cents

great work, thanks again!

Noisyfox, 2024-01-26:
> i just happened to have a need for this and found it merged into the nightly orca build. i have about 50 objects on my plate and it seemed to get the order correct.
>
> if youre still searching for [...]

I'm not a native English speaker but "Intra-Layer Order" sounds just about right to me. Thanks!

NevesLF, 2024-09-12:
Hey, could we have a quick way to number each individual object on the plate when we clone them? It's great that we can set the print order by the object list, but when you have 100+ cloned objects with the same name, it's a pain to manually order them.

Maybe change it to a "1- Object"; "2- Object" etc. naming scheme in the object list when we clone it would help.

PR #3685 (2024-01-21): Allow specifying print order within each layer
Fixes #3555

Now you can drag the object list to reorder the object in "by layer" mode. And if you choose "as object list" option in the layer order drop down, then for each layer, the objects will be printed in that order.

Currently this only applies to object, not their parts. So it currently won't print the object parts use the order as in object list. Will try to improve this later.

Also, the tool change order has higher priority than object list order, and the object list order will be used within each color. For example, if you have object [... cut]