OrcaSlicer/OrcaSlicer #3556: V1.9 Support being printed as a solid (100% infill)

lin-ycv, 2024-01-08:
[image]

lin-ycv, 2024-01-08:
Same issue as #1303

lin-ycv, 2024-01-08:
works fine if the tree type is default, but default doesn't work with adaptive layer height

Noisyfox, 2024-01-12:
Could you upload the project?

lin-ycv, 2024-01-13:
Here's the project file
[SolidSupport.zip](https://github.com/SoftFever/OrcaSlicer/files/13926400/SolidSupport.zip)

Noisyfox, 2024-01-14:
Change base pattern to Default, or increase the Base pattern spacing to some value like 1mm.
[image]

lin-ycv, 2024-01-14:
Cant use default, because it doesn't support adaptive layer height, changing the base pattern spacing did the trick.

I think `Tree Strong` and `Hybrid` should also have a suggestion message like `Slim`, for changing the spacing
[image]

closed 2024-01-14 by lin-ycv