OrcaSlicer/OrcaSlicer #3564: Add Binary GCode Support .bgcode

warwickchapman, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
It will help with quicker uploads to PrusaConnect and downloads to the printer.

**Which printers will be beneficial to this feature**
All printers that support .bgcode

**Describe the solution you'd like**
Port .bgcode support from PrusaSlicer 2.7.0

**Describe alternatives you've considered**
I am unaware of alternatives.

**Additional context**
I did search to see if someone had requested this already.  Surprised not to see it.  I use it for all prints from PrusaSlicer.

Mikyner, 2024-01-13:
I also hope that bgcode support will be added. I would still appreciate direct upload to the print queue for Prusa connect.

kdobrev, 2024-05-12:
Why is there no activity on this topic? All youtubers say that all new things get implemented in OrcaSlicer, but it seems that the developers are only busy adding half backed profiles for every new 3d printer that comes out and not actually adding features to the slicer itself...

K1ngjulien, 2024-06-03:
Hello! I'd love to help implement this. Prusa Slicer uses [libbcgcode](https://github.com/prusa3d/libbgcode). maybe there's a way to cherrypick the commits over or just copy the implementation files

P-C-R, 2025-02-26:
+ again for support. More Core One users are using then orca (with printables intgegration)

BuxtonWater, 2025-03-16:
More support, again. This is a vital feature for MK4S and Core One users.

bwees, 2025-03-17:
I have created a post processor to make bgcode files as a solution until it gets native support:

https://github.com/bwees/orca_bgcode

WimVDK, 2025-06-23:
I'd love to help test this if/when this gets added

WimVDK, 2026-08-28:
Bumpetie bump. Still wanna seee this implemented