OrcaSlicer/OrcaSlicer #3566: support_critical_regions_only option not visible in slicer UI

OmegaModr, 2024-01-08:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Artilery sidewinder X2

### How to reproduce

1. Go to support
2. Enable supports
3. Option not available between "On build plate only" and "Remove small overhangs"

### Actual results

The option is missing, adding the option to the json profile appears to be working.
[image]

### Expected results

The option should be there

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13865287/log.zip)
[Midde_Finger_Pot_V2.zip](https://github.com/SoftFever/OrcaSlicer/files/13865288/Midde_Finger_Pot_V2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2024-01-12:
This option is not available for default/organic tree style.

OmegaModr, 2024-01-12:
Mea culpa, was not aware of that fact - can disregard this ticket then.

closed 2024-01-12 by OmegaModr