OrcaSlicer/OrcaSlicer #3576: Copy settings from one object to another - Simple, fundamental.

PhilBaz, 2024-01-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Yes, currently you can copy a set of object specific settings from one object to another.

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

All

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Copy and paste settings. Could be 'copy object settings' in right click menu, along with paste. Or the ability to drag an icon in the object list would be cool.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

You have to manually change them all. It sucks!

**Additional context**
Add any other context or screenshots about the feature request here.

This seems like a fundamental workflow shortcoming. This is a logical and necessary feature for Orca. :) Hail the Whale!

DARKFiB3R, 2024-01-24:
This would be so helpful right now.

Also, once you have made changes to the settings of a multi part object, you can no longer Reload from disk or Replace with STL, which sucks.

PhilBaz, 2024-04-24:
Bumping as this is fundamental functionality that is currently missing.

SamuraiBuddha, 2024-08-03:
just came across the need for this today, I second OP's suggestion!

Wang-Thunder, 2024-09-19:
Yup, bump.

lukasmalkmus, 2024-09-27:
Yes, just wanted to do something like this today, would be very helpful.

Thegribbs, 2024-10-04:
Bump. this would be an amazing feature to have in the the 'view all objects settings pup-up menu.

brysonhyte, 2024-11-04:
The desired behavior is achievable by selecting the part you want to clone from first, then selecting the parts you want to clone the settings to.  Once they are all selected, right-click on the main part and select Assemble.  Once the parts are grouped into an assembly, you can ungroup them by right-clicking and selecting Split->To Objects.

carlosjln, 2024-11-12:
Indeed, it would be quite helpful for us folks who work with repeated settings on different STLs, since there isn't a way to define a process that enables height range modifiers and all settings while just swapping the object.

The menu options could be inserted after "Height Range Modifier" as "Copy Settings" and "Paste Settings",

[image]

DARKFiB3R, 2024-11-15:
Just wanted to say thanks again to [brysonhyte](https://github.com/brysonhyte)

The workaround has been so helpful while making loads of tiny modifications to the model I'm working on, while I try to really dial it in for the print process. I'd usually be pulling my hair out by this point.

I suppose saving my current settings as a profile would help, but then I'm also constantly messing with slicer settings too, and I don't want a ton of profiles cluttering up the list.

emutier, 2025-01-08:
my use case: temperature calibration on linux. 
the stl is broken so I wanted to I fixed in an online tool. however, there is no way copy the object settings settings from the calibration stl

holser, 2025-03-01:
Alternatively, there should be a clone mode. In that mode when user edits one object the changes will apply to all "clones". That will help users when they need to adjust object later when all clones are placed on plate.

zombodotcom, 2025-04-05:
how is this not a thing still

Yankeedoody, 2025-05-25:
Bump! A "copy object settings" button would be incredibly useful!

Andreuha, 2025-08-08:
Bump... came here looking for a way to apply settings from one object in a project to another, so sad this is impossible. Or let me save an object's settings as a preset.

DARKFiB3R, 2025-08-09:
> Bump... came here looking for a way to apply settings from one object in a project to another, so sad this is impossible. Or let me save an object's settings as a preset.

Soon. It's already available in BambuStudio.

1NEGROUP, 2025-12-03:
Bump feature needed