OrcaSlicer/OrcaSlicer #3579: Painting and manual suport drawing makes object invisible

SeriuxExe, 2024-01-09:
### OrcaSlicer Version

1.9.0

### OS version

windows 11

### Printer

Ender 3

### How to reproduce

press "L"

### Actual results

object and build plate becomes invisible

### Expected results

not become invisible

### Project file & Debug log uploads

///

### Checklist of files to include

- [X] Log file
- [X] Project file

Thismadara, 2024-01-10:
Hello, same problem with flsun v400

mobiobione, 2024-01-10:
Same here, X1C.

bikerman51, 2024-01-12:
Same problem with my win7 desktop, even with the portable edition, but I don't have this issue on my laptop (win 10pro).

ryanaukes, 2024-01-24:
I am having the same problem. 
Orca Slicer version 1.9.0 running on my laptop (Windows 10 Pro 22H2, AMD Ryzen 7 4800U processor with built in Radeon Graphics)

closed 2024-05-01 by github-actions[bot]