OrcaSlicer/OrcaSlicer #3581: Bug with round table

Maksimius, 2024-01-09:
### OrcaSlicer Version

V1.9.0

### OS version

Windows 11

### Additional system information

CPU: Intel core i7
GPU: Nvidia FeeForce 2Gb
Mem: 32Gb

### Printer

Custom delta

### How to reproduce

1. Go to printer settings
2. Select table shape: Circular
3. Click Ok
[image]

### Actual results

[image]

### Expected results

Normal table view

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13875205/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

tastechnician, 2024-01-10:
OrcaSlicer Version
V1.9.0

OS version
Windows 10

CPU: AMD Ryzen 7 1700
GPU: AMD RX580
Mem: 16GB

Issue does not occur in previous versions, but occurs with 1.9.0.

PeterBeklemishev, 2024-01-27:
Same on nightly build.

metatron-0, 2024-02-14:
This problem is eliminated by choosing a model for the table. You can simply create a plane of the desired diameter.
[image]
[image]

closed 2024-07-10 by github-actions[bot]

jmezzacappa, 2024-10-18:
This bug still exists in OrcaSlicer 2.1.0. Should this issue be reopened, or should I open a new one?

I tried loading a custom model for the bed, but that only solves the "cosmetic" problem (the appearance of side-by-side beds). The "arrange all objects" feature is still broken.

This is where the "arrange all objects" feature places the models:
[image]
(OrcaSlicer 2.1.0, macOS Sonoma 14.6.1, Apple M1 Pro CPU)

TexZeTech, 2024-10-29:
> This bug still exists in OrcaSlicer 2.1.0. Should this issue be reopened, or should I open a new one?
> 
> I tried loading a custom model for the bed, but that only solves the "cosmetic" problem (the appearance of side-by-side beds). The "arrange all objects" feature is still broken.
> 
> This is where the "arrange all objects"  (OrcaSlicer 2.1.0, macOS Sonoma 14.6.1, Apple M1 Pro CPU)

I just tried to replicate the bug that was in this bug report and could not replicate in v2.2.0-rc on Linux.

The issue you are talking about "arrange all objects" should have its own separate bug report.

I would also upload the part file you are using, log files & even a project file for others to try and replicate.

RandomGgames, 2025-01-05:
Still an issue in v2.2.0

charlesmeyer, 2025-01-19:
Can also confirm this bug still exists.

bruce98101, 2025-03-28:
Been following this for a while and updating that it's still a bug with OrcaSlicer_Linux_AppImage_V2.3.0.AppImage.