OrcaSlicer/OrcaSlicer #3582: Timelapse prompt disappeared on print confirmation modal in a printer

Voltmod, 2024-01-09:
### OrcaSlicer Version

1.9.0

### OS version

macOS 13.6.1

### Printer

Bambu Lab X1C

### How to reproduce

1. Bambu X1C with microSD card
2. Slice any model for it using OrcaSlicer 1.9.0
3. Send it to the printer using the "Send" button
4. Manually on the printer screen - select the model to print and observe the absence of the option to enable/disable timelapse

### Actual results

Timelapse is enabled all the time without the possibility to disable it.

### Expected results

Timelaps prompt (check mark) appeared in the modal window of printing details confirmation. 

### Project file & Debug log uploads

[timelapse-issue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13875256/timelapse-issue.3mf.zip)
[timelaps-issue-log.zip](https://github.com/SoftFever/OrcaSlicer/files/13875257/timelaps-issue-log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bbbrown70, 2024-01-12:
can confirm and replicate this. without the checkbox on the printer timelapse is **always** enabled and recorded for the model regardless of what was set in orca when sent

closed 2024-04-20 by github-actions[bot]

ccatlett1984, 2024-06-21:
Still in issue in latest build ( 2.1.0)